gnea/grbl #320: override Z depth while running

85mattia, 2017-12-11:
Hi All
it is possible to override Z depth cutting white code is running ?
Sometimes i need to correct the depth on the fly to trim the cut.
...There is a trick to make this ?

X3msnake, 2017-12-11:
Only if it is implemented on one of the gcode senders.

It would be a case of adding a offset. Before sending it to grbl

Dont know if any current sender has this implemented. Maybe bCNC

85mattia, 2017-12-11:
I use UGSplatform, but i don't think implements this feature

X3msnake, 2017-12-11:
Do you know any programming?

You can ask the ugs git for the implementation or the bcnc Git. They are
both very active hubs.

I use bcnc it is written in python and is a pluggable structure so one cab
write its own plugin to use with it.

Post back if you manage a solution. Otherwise I can have a look at it and
see if i can help you implemented this feature for the community.

85mattia, 2017-12-11:
thanks for the suggest
Now ugsplatform have plugins. 
 I know something programming, but before i can ask if someone have develop something similar.
thanks

closed 2017-12-11 by chamnit