gnea/grbl #323: feature request: After homing, print how far off the position was, like probing

myself248, 2017-12-12:
To tune a machine and check for missed steps, I've been setting up a dial indicator, doing a bunch of moves at various feed rates, and coming back to the dial indicator.

Which is nice and precise, but suppose I don't feel like getting out the dial indicator. The machine already has switches on it.... why not use those?

I'd love to have a [PRB: x.xxx y.yyy z.zzz] style output of where the machine THOUGHT it was when it finally hit the home switches. Of course it would ideally always read zeroes...

Alternately, this request could be thought of as "Let me do a probe cycle against each home switch", and use all the normal probing code, just change which pin is being sensed. If there's already a way to do this, I apologize for not finding it in the docs.

mpatoulachik, 2018-03-05:
I agree, this is a very nice feature to have (I have it with Galaad, and can confirm...)
Sometimes you miss only a handful of steps and can't figure why your machining isn't quite right. It also really helps when tuning up speed/acceleration.

109JB, 2018-03-06:
https://github.com/grbl/grbl/issues/814

petmakris, 2018-03-23:
Shouldn't this be a feature of a controlling IDE? Such as UGS?

myself248, 2018-03-23:
@petmakris How would the sender get the data if the machine doesn't report it somehow? Yes, yes it should be.

petmakris, 2018-03-23:
My bad, I read your post quickly.

langwadt, 2018-03-23:
I think if you change the homing to do one axis at a time you can do it with one line of code, copy the sys_position[axis] to sys_probe_position[axis] before setting the home position in limits.c (line ~329)

sys_probe_position can be queried with $#