gnea/grbl #325: What are hard and soft requirements for the pinout?

oyvindkinsey, 2017-12-14:
I'm building a hybrid CNC/3D Printer using the Arduino Nano with grbl and Teacup as respective firmwares. These have different pinouts, so my plan was to build a custom shield and modify the source of both firmwares to use a common pinout scheme.

To accommodate this, I also need to use analog pins for the limit switches, since I need access to two more digital ports for the extra stepper motor, as well as one more PWM  port for the extra heater (will use spindle PWM to control hotend heater).

Now, while I do have a design in place, I haven't actually tested this, and I just realized that it's mentioned in various places that grbl requires all the step pins to be on the same port etc.

Would it be possible to get documented what hard requirements exists, and what we can easily configure in code?

chamnit, 2017-12-28:
It has to do with the pin change interrupts. They operate per PORT and provides no information on which pin triggered or how. Grbl does not check the pin state within a pin change interrupt, because there is a very small chance it can be misread. For hard limits and user safety, Grbl interprets a pin change as a hard limit event.

chamnit, 2017-12-28:
Also the direction pins are on the same port because it's easier and faster to setup the IO in the stepper interrupt with one 8-bit variable. If they were on different ports, there would need to be a lot more code to set these various pins in an ISR that is very time sensitive.

closed 2017-12-28 by chamnit

chamnit, 2018-01-22:
I suppose, but it hasn't been a problem given the number of shields available for Grbl these days. The Grbl project was intentionally designed to simple to facilitate the advanced parts of its development, i.e. realtime overrides. It's never been advertised to be a complete solution for everything. Don't worry, it's all part of a grand plan.