gnea/grbl #327: Status report request (?) sends 'ok' as respond

Dani88, 2017-12-15:
Based on the latest documentation [here ](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface): 

> Push messages are easily identified because they **don't start with an ok or error** like response messages do. They are typically placed in [] brackets, <> chevrons, start with a $, or a specific string of text.

When: I send a `?` command (real-time status report request) to the GRBL

**Expected**
Then: Grbl should response with the status report (between chevrons) - as the the documentation states

**Actual**
Then: Grbl response with the status report AND send back an `ok` message as well as two separated line.

Let's say I'm implementing basic streaming protocol for GRBL, which waits for `ok` response before a new command will be sent. Theoretically the `?` request is a big exception since I can send it with 200 ms delays over and over again, regardless of other commands, or at least that's what the documentation says under the 'Status Reporting' section:

>The ? is always picked-off and removed from the serial receive buffer whenever Grbl detects one. So, these can be sent at any time. Also, to make it a little easier for GUIs to pick up on status reports, they are always encased by <> chevrons.

If I send a command to GRBL (for eg.: I'm streaming a *.nc file line-by-line) I'll have to wait for an `ok` response between every lines, but a polling solution is also running in the background and sending `?` requests to GRBL on another thread with low frequency. I can't really match the request-response pairs, because the very same response will be sent to me via GRBL. The next `ok` may be the matching pair for my status report request or for the last line from the *.nc file.

It looks like I should suspend the polling everytime when I'm waiting for an `ok` response, but the documentation is not very clear in this topic, also makes false statements.

chamnit, 2017-12-15:
Make sure you aren't sending a carriage return or line feed with your '?' character.

closed 2017-12-15 by chamnit

Dani88, 2017-12-15:
I'll never tell anyone how much time did I spent with this issue :)

Thanks for your quick answer, works perfectly!