gnea/grbl #328: When [Connect] is completed, there is an error that red [HOME] and Overrides are missing!

viewsat, 2017-12-18:
When [Connect] is completed, there is an error that red [HOME] and Overrides are missing, but [HOME] and Overrides will appear after [Get Grbl Settings] is pressed.

My cnc machine has a home limit switch.

I use the candle and ugsplatform G-Code-Sender to do the test, when [Connect] is completed, [Home] and Overirdes are not wrong.

<1> [HOME] and Overrides are missing
[image]

<2>[HOME] and Overrides will appear after [Get Grbl Settings] is pressed.
[image]

chamnit, 2017-12-18:
@viewsat : This is a GUI issue. Not a Grbl firmware issue. Please post in the appropriate place.

closed 2017-12-18 by chamnit

viewsat, 2017-12-19:
This is a GUI issue,My problems have been solved.