gnea/grbl #329: Serial behavior

ilie321, 2017-12-20:
Hello i notice that compared to 0.9j if i turn off power and after some time i do power on it not get in consideration the commands from pc only if i send reset command. It is possible to adjust?  Thank you.

chamnit, 2017-12-20:
No. There has been no changes on how Grbl connects. This is from the Arduino boot loader and your computer.

closed 2017-12-20 by chamnit

ilie321, 2017-12-20:
I used same bootloader for 0.9j and 1.1f and the same software . this is why i suspected something in  grbl because of new real time commands or idk

chamnit, 2017-12-20:
Grbls comm protocol has changed, as outlined in the wiki. Not the serial handling or how it operates as a whole.  Your GUI may have two different ways of handling each version. This may be what you are seeing as a difference.

ilie321, 2017-12-20:
Thanks for quick answers i will check.  I use candle. I have used for tests the old one for 0.9 which can't identify 1.1 but it sends the commands .

biasedlogic, 2017-12-21:
I'm personally on a fence with candle. I love some parts of how it's designed, but it's bug ridden in the serial department. It seems machine dependant, i have a laptop where it just works and i have a PC where candle will skip gcode commands and end up crashing the complete computer down to bsod. Long story short, try another sender

ilie321, 2017-12-21:
For me it works great except autoleveling

biasedlogic, 2017-12-21:
Except *"Hello i notice that compared to 0.9j if i turn off power and after some time i do power on it not get in consideration the commands from pc only if i send reset command."* - i mean, do try other sender!

ilie321, 2017-12-21:
I did with other 2 same result. Same bootloader. On 0.9j i get connected after power off power on in few seconds.  On 1.1 i need to press the reset from software to get connected again.  But is not  big issue i will live with it

biasedlogic, 2017-12-21:
A long shot: it sounds like the reset circuit of your Arduino might not get reliably reinitialised between power down and power up. To some extent it depends on the configuration (number of active pull ups and active outputs).

ilie321, 2017-12-23:
On 1.1 it should be the short noise at power up from motors?