gnea/grbl #331: Unexpected full circle motion

pzobolyak, 2017-12-21:
Hi,

During cutting I experience an unexpected motion that cuts into material off the calculated toolpath.
Reloading the g code file into my editor shows no problem in the toolpath, but the actual cut behaves weird.

https://photos.app.goo.gl/lKfWpHnDgDSKcuS82

Attached a video where it is visible at 0:06

The expected path should be as the attached image shows.
I'm using 1.1j

[image]

neilferreri, 2017-12-21:
Can you share the gcode? What are you using to generate your toolpaths? What sender are you using?

jahnj0584, 2017-12-21:
Mostikely the wrong post processor arc settings or a stay down distance

chamnit, 2017-12-21:
It’s your post or CAM. Your file has tons of very very small G2/3 arcs. These commands are very sensitive to calculation and rounding errors due to how the arc gcode itself is mathematically posed. Grbl has a fairly lax error check for arcs but even then it will generate an error with some arcs time to time when it gets really bad. When this happens, a GUI can sometimes keep streaming and continue on the wrong path. It should stop upon error and is usually a preference option. 

To fix, don’t use arcs. Usually CAM posts can generate G1 commands only. (For precisely this arc issue). Sometimes you can specify the minimum arc length, if so, specify something larger than 1mm or so.

closed 2017-12-21 by chamnit

pzobolyak, 2017-12-22:
Setting the minimum arc length did help, thank you!