gnea/grbl #332: Step generation accuracy - worst case scenarios?

xenovacivus, 2017-12-23:
Hi,

I've been doing a bit of research into the accuracy of step profiles generated by stepper motor controllers like grbl.  I found a few cases where the width of steps is +-1/3rd of the ideal step, see here for details:

https://xenovacivus.github.io/jekyll/2017/12/23/stepper-motor-driver-accuracy.html

The short/long steps clearly aren't an issue, otherwise there would be some complaints.  They average out pretty quickly - especially with microstepping.  Has anyone else done testing to see what the absolute worst case is though?  And is the state-of-the-art approach for testing step profiles still "Run some GCode and see how the machine performs."?

chamnit, 2017-12-23:
Good analysis. Your conclusions are correct. There are velocity spikes but they average out and are high frequency, which means they don’t excite the structure. Just slightly audible. The Bresenham algorithm is the primary culprit but is managed and reduced  with Grbl’s AMASS smoothing algorithm. It’s difficult to guarantee multiple axes to start and end simultaneously and remain synced if you have them independently timed. This is the reason for using Bresenham.  

Other velocity spikes is due to servicing other interrupts. They can intermittently run during a pulse and delay the start or end of one. Nothing I can do about that with the 328p. But should be eliminated with the upcoming faster ARM version.

closed 2017-12-23 by chamnit

xenovacivus, 2017-12-24:
Yeah, short of polling for everything in the main loop (and requiring flow control on the USART), there's not really a way to get around having other interrupts in the AVR land.  The ARM Cortex M3/M4 have this cool thing called the NVIC which allows an interrupt to interrupt another interrupt based on priority.  So you can run the step code in the highest priority interrupt and ensure it won't be disturbed by other tasks.

The downside to the ARM devices is interrupt latency and jitter - the time to service an interrupt after the flag is set (12-16 cycles + others depending on state) is less consistent than AVR, which might reduce step accuracy.  But with the higher frequency (120Mhz+), the jitter probably won't have any noticeable effect.  +- 6 cycles would be ~10Mhz - and the chopper drive in the stepper motor driver would be running about two orders of magnitude slower than this, so it's very unlikely that noise would propagate into the rest of the system.