gnea/grbl #334: How to set Homing on boot?

TehseenHasan, 2017-12-27:
On my DIY laser engraver, I want to home the machine every time I restart or turn on my laser machine. How can I add this option in GRBL code or is there any predefined option present in GRBL firmware? 
Thanks!

zeevy, 2017-12-27:
You can set $N0=$H

TehseenHasan, 2017-12-27:
It gives error message : `Expected command letter`

zeevy, 2017-12-27:
If not there, may be its not possible, GRBL executes those two startup blocks every time it starts

chamnit, 2017-12-27:
Automatic homing is explicitly prevented due to the high probability of crashes. Do not program the startup lines with motion commands. It is very unsafe and will cause your machine to move unexpectedly. You need to maintain control of the machine at all times to ensure personal safety.

closed 2017-12-27 by chamnit