gnea/grbl #335: bCNC  "ALARM:3" status makes me crazy.

Kozze, 2017-12-27:
Hi everyone,

Beginning 2017 I have built my own CNC.

At the beginning, everything was running fine without limit switches.
After installing limit switches, I became almost by every run an alarm, this problem was clear and fixed after installing opto couplers between the switches and the Arduino.

Now, we are several months further and time to time the CNC stops without a visible reason.
In bCNC, the “ALARM: 3” status shows up.

Not good… the manual says "Reset while in motion". 
-	Grbl cannot guarantee position. 
-	Lost steps are likely. 
-	Re-homing is highly recommended.

I cannot find the reason why this behavior occurs, and where it comes from.
I hope that someone here can give me a tip where I can start searching to fix this problem for good.

In the meantime, at the end of the motor cables (as close as possible to the Arduino), like also on the power cable of the Dewalt 611 router, I installed ferrite suppression filters, unfortunately without any result, the machine goes occasionally into alarm phase 3.
 
1.	Where can I start to search what is happening here?
2.	I cannot find log files of GRBL or bCNC, where are they stored??

Any help is welcome and much appreciated !

Best regards,
Benny

chamnit, 2017-12-27:
It sounds like you are getting some electrical noise on your A0 soft reset pin. You can probably put a capacitor there to help level out the noise.

Kozze, 2017-12-27:
Hi chamnit,

That is a possibility.

What kind of capacitor should I use and how should I connect it ?
Is there any electrical drawing/plan that I can use?

Regards, Benny

chamnit, 2017-12-28:
I think just about any small capacitor should work wired across the A0 pin and ground, but it would depend on if A0 is held high or not. It should be from what I recall.

It could also be related to a bad ground or ground loops too. I've heard of a lot of situations in the past where placing things like the spindle and fluorescent lights on a different circuit as the controller or star grounding everything will eliminate weird issues like this.

Just proceed with caution. I'm not a electrical engineer or a hardware enthusiast. So take my advice with a grain of salt.

Kozze, 2017-12-28:
OK chamnit,

I will first try with a 0,1 µF capacitor between the A0 and the ground. 

(Today I have carved multiple toolpaths  for at least 2 hr. without any Alarm)

I thinking about to buy the” Arduino CNC Isolation Shield”… there the feed hold, probe and reset/abort lines are also connected with a opto isolated inputs. 
[http://www.yrcnc.com/product/arduino-cnc-isolation-shield/ ](url) any comments about this board?

Regards,
Benny

Stephan79, 2018-01-23:
Il semblerait que le parasitage provient d'ailleurs que des FdC... Mes entrées FdC sont opto couplées et de temps à autres, j'ai effectivement des erreurs 3 sur bCNC...
En écartant la CNC d'un tube fluorescent, je n'ai plus ces erreurs...
Steph

Kozze, 2018-01-23:
FYI
Since I put the 0,1 µF capacitor, I didn't have "Alarm 3" errors anymore. 
I hope it will stay like this.
Benny

ghartc, 2018-01-23:
Hate to be rude, but the off topic stuff is bad enough without seeing this
"merde" (crap).

Stephan79, 2018-01-23:
@ghartc :  Ta grossièreté à l'air de bien représenter ce que tu es :)
C'est toi qui est hors sujet, patate !!

109JB, 2018-01-23:
> Hate to be rude, but the off topic stuff is bad enough without seeing this
"merde" (crap).

You say you hate to be rude, but then just went ahead and were rude anyway.

closed 2018-02-02 by chamnit