gnea/grbl #336: Update Woodpecker CNC GRBL 0.9 to 1.1f

fxh1357, 2017-12-28:
Hello, I just recently decided to update my Chinese “Woodpecker CNC Board” to the most up-to-date grbl Version 1.1f. 

I found [Mariano Gaston’s YouTube video](https://www.youtube.com/watch?v=kJRy3_heliQ) that promises the feasibility of this procedure and shows the necessary connections.  I also read the issue about connecting limit switches to this board (https://github.com/gnea/grbl/issues/123)  and as mentioned there I re-checked all pins from the header regarding there mapping to the 328P. For some reason I can’t get a direct connection between the pin associated with MISO and its counterpart on the chip itself. The connection seems to be bridged by something – maybe a diode? Nevertheless, I tried to connect my programmer (usbasp-clone and ArduinoISP). - Long story short, after several ours of fiddling around I finally managed to brick my controller board.

I know this is most probably not an issue with GRBL itself and thus should not be posted as an issue for that, but I have no idea where to look for support regarding this specific problem. So sorry for that, but any help is highly appreciated!

Following first what I’ve done to kill the board and then what I’ve tried to get it back up.

As described above I am using an [usbasp-clone](https://de.aliexpress.com/item/USBasp-USB-ISP-3-3V-5V-AVR-Programmer-USB-ATMEGA8-ATMEGA128-New-10PIN-Wire-Support-Win7/32536193605.html) and an [Arduino Nano as ArduinoISP](http://www.martyncurrey.com/arduino-nano-as-an-isp-programmer/). From the begin on I had problems connecting the Woodpecker using the usbasp-clone, so I mostly tried to update with the avrisp. With all connection in place and the external power supply connected I got something like this from avrdude:

`$ avrdude -c avrisp -p m328p -P com6 -b 19200 -B 100

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude.exe: Device signature = 0x0fca07
avrdude.exe: Expected signature for ATmega328P is 1E 95 0F
             Double check chip, or use -F to override this check.

avrdude.exe done.  Thank you.`

After several hours I thought it would be a great idea to just modify Arduino’s avrdude configuration to expect this device signature. I clicked the “Burn bootloader” entry, it did something and then crashed. I’ve got no log from that procedure but now, the board is not serial-printing its version number on startup anymore.

Since then I’ve tried many different configurations, with and without external power supply using windows and linux hosts, testing my programmers with other 328p chips, etc. 
My current test is just an avrdude invocation to query the device signature. Both programmers are working, I verified that with the other chip. For that reason, I assume the programmer-side wiring is correct as well. I checked the wiring to the Woodpecker board multiple times and for the MISO pin that is connected via an unknown component I also tried to bypass it by connecting the wire directly to the atmega pin.
Using the usbasp-clone I always get an ‘initialization failed’ (windows and linux host). With the ArduinoISP I get two different responses depending on the attachment of the programmers VCC line - despite the external supply – It’s either `0x0fca07` (with connected VCC) or `0x1f9f0f` (with disconnected VCC) but never 0x1e950f what it should be. I tried different values for the -B parameter without any success.

So, any suggestions?

chamnit, 2017-12-28:
From what I can glean from your post, it doesn't look good for you. That youtube video said nothing about the specifics of burning v1.1, only that its possible. It also didn't say anything about having to alter the Arduino config to get it to work. 

In my experience, I've never had any problems with flashing via programmer. I just made sure that everything was disconnected from the controller, since the programming pins are shared on the 328p. I also just use command line to compile and flash. I also often flash with the '-F' to override the signature check, which may or may not help you.

Anyhow, closing this issue since it's not a Grbl software problem, but you and others can still comment after.

closed 2017-12-28 by chamnit

fxh1357, 2017-12-29:
Thank you so much for your quick response. "[...], often with the '-F' to override" and the fact that my board was already bricked made me confident in retrying the update procedure and it just worked ;)

I used the Arduino IDE, an external power supply, disconnected stepper motors and spindle. I connected an Arduino Nano as described in my previous post and changed the device signature in the avrdude.conf

```
 RST       MISO
 |SCK      |MOSI
ooooooooooooooooo
ooooooooooooooooo
GND
```

FelisucoVFR, 2018-01-11:
Hi fxh1357!
Good success!

I don't know how to start with! Can you please help me?
May I know a step by step procedure? Is it possible taking the actual .hex as a back up?
Do you create the .hex with IDE and then upload it?
Do you have any picture of the connections from your Nano to the Woodpecker board?
Which selections at the IDE? "Arduino as ISP"?
External power supply. Where? 5v?

Thanks!

VirtualEnder, 2018-01-13:
So I'm a complete idiot and forgot to copy the settings out of grbl before I flashed this thing.  Did anyone happen to save those settings, or have the correct settings for this unit? 

On another note, my board had the Arduino bootloader already installed and flashes fine via the Arduino software. I just had to select Arduino Uno as the board and and it worked flawlessly.  Now if I could just get the settings back....

biasedlogic, 2018-01-13:
Well, your settings depend on your machine... Not knowing what that is will make it hard to come up with reasonable values.
On the other hand, the only thing set right in the original settings of my 1610 was the step per mm setting... It may be better to just start from scratch anyway

X3msnake, 2018-01-13:
You can always reflash the original firmware from your machine and run a
serial connect and call the settings and save it in txt

then reflash it

fxh1357, 2018-01-13:
@X3msnake Can you provide a link where to get the 'original firmware'? I think that would be quite difficult in respect to the variety of ebay and AliExpress sellers using slighltly different Boards and components -

fxh1357, 2018-01-13:
@VirtualEnder find following the settings I am currently using -

```
#ifdef DEFAULTS_MINI_1610
  // Grbl generic default settings merged with the values present in the MINI 1610 CNC stock firmware (GRBL 0.9 on a WOODPECKER CNC board).
  #define DEFAULT_X_STEPS_PER_MM 800.0
  #define DEFAULT_Y_STEPS_PER_MM 800.0
  #define DEFAULT_Z_STEPS_PER_MM 800.0
  #define DEFAULT_X_MAX_RATE 5000.0 // mm/min
  #define DEFAULT_Y_MAX_RATE 5000.0 // mm/min
  #define DEFAULT_Z_MAX_RATE 800.0 // mm/min
  #define DEFAULT_X_ACCELERATION (10.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_Y_ACCELERATION (10.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_Z_ACCELERATION (10.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_X_MAX_TRAVEL 200.0 // mm NOTE: Must be a positive value.
  #define DEFAULT_Y_MAX_TRAVEL 200.0 // mm NOTE: Must be a positive value.
  #define DEFAULT_Z_MAX_TRAVEL 200.0 // mm NOTE: Must be a positive value.
  #define DEFAULT_SPINDLE_RPM_MAX 7000.0 // rpm
  #define DEFAULT_SPINDLE_RPM_MIN 0.0 // rpm
  #define DEFAULT_STEP_PULSE_MICROSECONDS 10
  #define DEFAULT_STEPPING_INVERT_MASK 0
  #define DEFAULT_DIRECTION_INVERT_MASK 5
  #define DEFAULT_STEPPER_IDLE_LOCK_TIME 25 // msec (0-254, 255 keeps steppers enabled)
  #define DEFAULT_STATUS_REPORT_MASK 3 // MPos enabled
  #define DEFAULT_JUNCTION_DEVIATION 0.01 // mm
  #define DEFAULT_ARC_TOLERANCE 0.002 // mm
  #define DEFAULT_REPORT_INCHES 0 // false
  #define DEFAULT_INVERT_ST_ENABLE 0 // false
  #define DEFAULT_INVERT_LIMIT_PINS 0 // false
  #define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
  #define DEFAULT_HARD_LIMIT_ENABLE 0  // false
  #define DEFAULT_INVERT_PROBE_PIN 0 // false
  #define DEFAULT_LASER_MODE 0 // false
  #define DEFAULT_HOMING_ENABLE 0  // false
  #define DEFAULT_HOMING_DIR_MASK 0 // move positive dir
  #define DEFAULT_HOMING_FEED_RATE 25.0 // mm/min
  #define DEFAULT_HOMING_SEEK_RATE 500.0 // mm/min
  #define DEFAULT_HOMING_DEBOUNCE_DELAY 250 // msec (0-65k)
  #define DEFAULT_HOMING_PULLOFF 1.0 // mm
#endif
```
Concrete changes are the steps_per_mm, max_rate, spindle_rpm_max, direction_invert_mask and status_report_mask

fxh1357, 2018-01-13:
@FelisucoVFR 

> May I know a step by step procedure? Is it possible taking the actual .hex as a back up?

Have a look at the guide provided in the [grbl wiki](https://github.com/grbl/grbl/wiki/Compiling-Grbl). To get the settings right you need to also check the "Advanced Users" section to alter the configuration as I mentioned in my last comment. 
To use an Arduino as a Programmer there is a guide as well, see [Flashing Grbl to an Arduino](https://github.com/grbl/grbl/wiki/Flashing-Grbl-to-an-Arduino)
To use an Arduino Nano see [Arduino Nano as an ISP Programmer](http://www.martyncurrey.com/arduino-nano-as-an-isp-programmer/)

On top of that you will need to adjust the expected device signature to trick Arduino IDE to flash the firmware regardless of the failed device signature check. (I also removed all cables running to the board and the stepper drivers to minimize side effects.)
To change the signature:
- Try to upload the sketch, avrdude will print out the expected and actual values. 
- Copy that actual device signature and put in in your avrdude.conf at the section describing the m328p (`id = “m328p”`). Just add a # in front of the original device signature to easlily restore it afterwards an add a new line like `signature = 0x0f 0xca 0x07;` The file can be found at `%localappdata%\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9\etc\avrdude.conf`
- You should now be able to flash the firmware.

I had some changes in the read device signature during my testing, so maybe it fails on your first attempts. Just retry, or change the signature again. **Be aware, that avrdude will not be able to verify the written firmware, most likely due to the same reason that changes the read signature! So an error in the console after uploading is expected.** 

I don't think, that it is possible to get the currently persisted *.hex file out there because some peripherals are seeminlgly interfering the downstream, thus the wrong device signature and failing checksum. I could be wrong at that point. The *.hex file itself would not get you anywhere because the problem itself is getting it onto the board.

> Do you create the .hex with IDE and then upload it?

Arduino IDE creates the *.hex file on the fly when uploading.

> Do you have any picture of the connections from your Nano to the Woodpecker board?

Have a look at [Connecting ArduinoISP to WOODPECKER CNC](https://imgur.com/a/AaNnH)

> Which selections at the IDE? "Arduino as ISP"?

Correct "Arduino as ISP" and make sure to use Sketch->Upload using Programmer. Otherwise you will just uplaod the firmware to your Programmer

> External power supply. Where? 5v?

As mentioned in my comment two weeks ago I was finally successful connecting the WOODPECKER CNC board to its 12V power supply. But a 5V connection from the arduino without connected power supply should be fine as well.

I spend several hour fiddling around to get my CNC running again. And there is some chance that you'll brick your controller. I would not recommend doing this update when you need your machine sooner as a new board would arrive in the mail.

X3msnake, 2018-01-13:
@fxh1357 

> Can you provide a link where to get the 'original firmware'? I think that would be quite difficult in respect to the variety of ebay and AliExpress sellers using slighltly different Boards and components

I cannot, don't know what the user board is neither the seller, if it was my board i would search for the specs and/or ask the seller for the original hex file to do reupload.

It might end up being too much hassle and I would probably just pull the datasheets on the motors and mesure the axel pitch and use the reprap calculators to find the closest settings for the machine i had at hands

https://www.prusaprinters.org/calculator/

VirtualEnder, 2018-01-13:
Thanks @fxh1357,

I already started rebuilding the config. I think we have basically the same machine, except mine is the 3018 version.  The direction inversion mask is the same, but I'm still fussing with the steps/mm. I got to 750, but oddly the Y axis seems to be slightly different from the X and Z axis. I need to have it cut some lines along each axis at a fixed depth and take some measurements, but it's pretty close. 

Unfortunately the board has been rebooting part way through the cut, which I'm still debugging. That is part of why I was trying to upgrade, but it hasn't solved the issue. Hopefully I can figure it out.

FelisucoVFR, 2018-01-13:
@fxh1357 
Thanks a lot! I’ll try soon!
Thanks for all the notes. It took me some time studying the Arduino as ISP but I catch it.
I did all as you described except for the signature as I used the -F option at avrdude. I just copied the long avrdude text the IDE wrote and added the -F in a cmd. It overrides the signature but got different error. I don’t remember now the error exactly, sorry.
I will update with better info about the fail I get.
I have the UNO I used as ISP and other CNC shield for it so I am not worry about the board but I would like to use this one.

Thanks again!

FelisucoVFR, 2018-01-14:
I cannot success @fxh1357 
Itryed many times again but no way. I am getting the following:

C:\Users\3Dsuco>C:\Users\XXX\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\XXX\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -cstk500v1 -PCOM7 -b19200 -Uflash:w:C:\Users\XXX\Documents\Arduino\Build_Folder/grbl11f.ino.ino.hex:i -F

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\XXX\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"

         Using Port                    : COM7
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.10s

avrdude: Device signature = 0xffffff (probably .reduced_core_tiny) (retrying)

Reading | ################################################## | 100% 0.08s

avrdude: Device signature = 0xffffff (probably .reduced_core_tiny) (retrying)

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0xffffff (probably .reduced_core_tiny)
avrdude: Yikes!  Invalid device signature.
avrdude: safemode: hfuse reads as FF
avrdude: safemode: efuse reads as FF
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "C:\Users\XXX\Documents\Arduino\Build_Folder/grbl11f.ino.ino.hex"
avrdude: writing flash (30580 bytes):

Writing | ################################################## | 100% 43.75s

avrdude: 30580 bytes of flash written
avrdude: verifying flash memory against C:\Users\XXX\Documents\Arduino\Build_Folder/grbl11f.ino.ino.hex:
avrdude: load data flash data from input file C:\Users\XXX\Documents\Arduino\Build_Folder/grbl11f.ino.ino.hex:
avrdude: input file C:\Users\3Dsuco\Documents\Arduino\Build_Folder/grbl11f.ino.ino.hex contains 30580 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 25.97s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0c
avrdude: verification error; content mismatch

avrdude: safemode: hfuse reads as FF
avrdude: safemode: efuse reads as FF
avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

Any idea?
Thanks

VirtualEnder, 2018-01-15:
So my rebooting issue was an issue with the computer not the board. I switched to my laptop and it works great! 800mm/s was dead on. Got some acceptable cuts out of 2mm and 3mm carbon fiber. Will try some 5mm later this week, but it seems as if everything is working! I think for a $300 amazon special, that's a good deal.

fxh1357, 2018-01-15:
@FelisucoVFR are you sure, that all your connections are correct?

FelisucoVFR, 2018-01-15:
@fxh1357 Quite sure as I did from 0 three times with different drawings. All of them matched except for the vcc which in one is disconnected. I tried anyway with same result.
Anyway, the avrdude makes the write, which can be unreal, but it reads later to check. I understand it works.
I did it without any motor nor pololu, with and without external power.
It seems a kind of write protect.

fxh1357, 2018-01-15:
@FelisucoVFR Do you always get the device signature 0xffffff?

FelisucoVFR, 2018-01-15:
@fxh1357
Correct. Always the same one.
Which Arduino board I should select? UNO, Nano, Nano pro, Mini, doesn’t matter?
Thanks a lot

FelisucoVFR, 2018-01-15:
@fxh1357
Well, I tried all. Just to save time next one not trying with all of them.

fxh1357, 2018-01-16:
@FelisucoVFR The chip on the woodpecker cnc is a 328p, so Nano should be right.

FelisucoVFR, 2018-01-17:
@fxh1357
More tests, same results.
My concern was the most about the autoleveling which is included with grbl 1.1f but I found the software OpenCNCPilot which works also on 0.9, so I think I am going to desist about this.
Many thanks for your help.

xseignard, 2018-03-01:
Hello,
Here is the command I used to upload grbl v1.1f to my woodpecker board (through the usb cable)

`avrdude -pm328p -carduino -P/dev/cu.wchusbserial1420 -b57600 -D -Uflash:w:grbl_v1.1f.20170801.hex`

Don't forget to save your settings before!! (running `$$`command)

akomelj, 2018-03-01:
I was also able to upload grbl v1.1f using xseignard's command to a stock Woodpecker 2418 CNC board via an Arduino Nano. I wired the Nano according to instructions above. I did it on Linux (serial port /dev/ttyUSB0) without changing any signatures in config files. I powered Woodpecker board from an Arduino 5V connection. Key thing for me was the baud rate (-b57600) - every other value reported some really strange errors.

JBSchueler, 2018-03-02:
Hi, I have a woodpecker PCB like this one and would like to update from 0.9 to 1.1f.
Does anyone known if this boards use the default grbl cpu pin mapping or do I have to changes/swap some pins?
[image]

JBSchueler, 2018-03-02:
Seems mine has a different baudrate (115k2) as others who where able to flash the woodpecker.

avrdude -CC:\Users\XXX\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -carduino -PCOM11 **-b115200** -Uflash:w:C:\projects\CNC\CNC3018\update_fw/grbl_v1.1f.20170801.hex:i

```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\XXX\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf"

         Using Port                    : COM11
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "C:\projects\CNC\CNC3018\update_fw/grbl_v1.1f.20170801.hex"
avrdude: writing flash (30492 bytes):

Writing | ################################################## | 100% 5.19s

avrdude: 30492 bytes of flash written
avrdude: verifying flash memory against C:\projects\CNC\CNC3018\update_fw/grbl_v1.1f.20170801.hex:
avrdude: load data flash data from input file C:\projects\CNC\CNC3018\update_fw/grbl_v1.1f.20170801.hex:
avrdude: input file C:\projects\CNC\CNC3018\update_fw/grbl_v1.1f.20170801.hex contains 30492 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.03s

avrdude: verifying ...
avrdude: 30492 bytes of flash verified

avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

C:\projects\arduino-1.8.5\hardware\tools\avr\bin>
```

akomelj, 2018-03-02:
TL;DR In the end, I flashed mine with 115200 as well (my board looks exactly like yours, JBSchueler).

I nearly earned myself a PhD from programming this beast. At first, I was getting some strange `not synced` and `response != expected` errors from avrdude trying to flash it directly using xseignard's command. Then I hooked it up via a spare Nano ("Arduino as ISP" type) and tried flashing again - and this time it worked. Well...

The only problems is that I forgot to change `-carduino` parameter of avrdude so I was flashing my Nano instead. :-) Imagine the surprise when opening serial connection and seeing grbl 0.9j greeting message after a "successful" flash. Once I figured that out, I changed the controller type to `-cavrisp` and successfully flashed it via Nano ISP (baud 19200). Just in case, I flashed stock Arduino Nano bootloader on top of that and later tried re-flashing it again directly without the Nano ISP (baud 115200) and this time it worked like charm.

Unfortunately, I still don't know wheather the initial errors were only temporary or not.

So, long story short for posterity: when flashing directly with avrdude, use `-carduino` and rates 57600 or 115200; when flashing with avrdude via Arduino ISP, use `-cavrisp` and rate 19200. If flashing the bootloader, first flash the bootloader and then grbl (with `avrdude -D` not to erase the bootloader).

JBSchueler, 2018-03-02:
Thanks for the update @akomelj, this is well explained.

I tried programming via an Arduino ISP programmer but was not able to succeed.
This is how I connected the SPI interface.
[image]

So I tried direct updating the FW via the USB connection of the woodpecker and seems it has a boot loader when shipped.

After flashed the woodpecker to 1.1f and send the configurations to the eeprom it works all just out of the box.

9a4gl, 2018-03-10:
Can anyone explain how to copy settings out before flashing to avoid problems that @VirtualEnder had ? In summary of 1.1 they mention EEPROM settings will be wiped and restored with new settings. I am new to this, what software tools you use to see/copy/restore those settings ?

JBSchueler, 2018-03-10:
Use `$$` to read the settings from the EEprom.
this can be done via any terminal program or most of the CNC programs have the option to enter manual instructions, like `$$`.
Copy and save the settings from the terminal into a file.
That file can be used to upload the old settings to the freshly burned board..

9a4gl, 2018-03-11:
Thanks. In that case I maybe have something for @VirtualEnder . I bought this https://www.ebay.com/itm/253468669130 it is 3018 with Woodpacker 2.8 board. Here are settings I read:
```
$0=10 (step pulse, usec)       
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool) 
$5=0 (limit pins invert, bool)  
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)     
$13=0 (report inches, bool)             
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)         
$22=0 (homing cycle, bool)                                                    
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min) 
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)   
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)         
$102=800.000 (z, step/mm)
$110=800.000 (x max rate, mm/min)
$111=800.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)              
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)                                               
ok             
```

nkakurai, 2018-03-13:
I managed to upgrade GRBL 0.9j to 1.1f on the apparently same board as the OP's and have just made this video:

"How to Upgrade GRBL 0.9j to 1.1f on Woodpecker CNC via Arduino as ISP (without soldering!)": https://youtu.be/noL3fKs0wxE

> I can’t get a direct connection between the pin associated with MISO and its counterpart on the chip itself. The connection seems to be bridged by something – maybe a diode?

I think it's MOSI (or at least in my case) and yes I wired Arduino's MOSI to one of the legs of the module IRF540 (connected to Woodpecker's MOSI) as shown in the video.
Hope this helps.

weiserlord, 2018-04-20:
Need Help :-(

I uploadet the bootloader successfully, the way nkakurai descript. But now i stuck with a blank Woodpacker because i am not able to upload the sketch. The promt shows: 

Der Sketch verwendet 30562 Bytes (99%) des Programmspeicherplatzes. Das Maximum sind 30720 Bytes.
Globale Variablen verwenden 1633 Bytes (79%) des dynamischen Speichers, 415 Bytes für lokale Variablen verbleiben. Das Maximum sind 2048 Bytes.
Wenig Arbeitsspeicher verfügbar, es können Stabilitätsprobleme auftreten.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x31
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x31
Problem beim Hochladen auf das Board. Hilfestellung dazu unter http://www.arduino.cc/en/Guide/Troubleshooting#upload.

The usb connection is runnig. Why it says that the sketch is to big? @nkakurai what showed your promt?

weiserlord, 2018-04-20:
Yehhaaa. I unpluged usb cable and short after i presst the reset button on the board. Then the Upload was succsessful

Cabalist, 2018-04-30:
I may have blown mine up trying to write out an arduino bootloader.  Could anyone post what there fuses are set to?  I foolishly flashed mine with arduino uno settings and now I am getting no response over serial.  

This command should print the fuse information to the console.  

`avrdude -c avrisp -p m328p -P /dev/cu.wchusbserial1410 -b 19200 -U lfuse:r:-:i -v`

Any help would be greatly appreciated.

CONSULitAS, 2018-05-06:
@Cabalist You gave me a nice riddle! 😄 

>I may have blown mine up trying to write out an arduino bootloader. Could anyone post what there fuses are set to? I foolishly flashed mine with arduino uno settings and now I am getting no response over serial.
>
>This command should print the fuse information to the console.

On my Mac i managed it with

```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude \
  -C /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf \
  -c avrisp -p m328p -P /dev/cu.wchusbserial146310 \
  -b 57600 -U lfuse:r:-:i -v 
```

Things you need to reproduce:
* baud rate 57600
* your usb port (depends on your hardware)
* paths to avrdude delivered with Arduino IDE
* connected via USB to Woodpecker GRBL 3.0.2 port directly (no USBASP or other programmer needed)

**Output**
```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/xxx/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.wchusbserial146310
         Using Programmer              : avrisp
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel AVR ISP
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading lfuse memory:

Reading | ################################################## | 100% 0.02s

avrdude: writing output file "<stdout>"
:0100000000FF
:00000001FF

avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

Any questions? You are welcome!

CONSULitAS, 2018-05-06:
BTW:
Perhaps the contents of the FLASH would help you, too?

Read with `-U flash:r:Desktop/woodpecker-grbl.hex:i`:
[woodpecker-grbl.hex.zip](https://github.com/gnea/grbl/files/1977527/woodpecker-grbl.hex.zip)

Read with `-U flash:r:Desktop/woodpecker-grbl.raw:r`:
[woodpecker-grbl.raw.zip](https://github.com/gnea/grbl/files/1977528/woodpecker-grbl.raw.zip)

Looks like everything worked fine reading the file (screenshot from raw file):
[image]

It is a GRBL 0.9j - see end of line starting with byte `1440`.

CONSULitAS, 2018-05-06:
BTW:
There are different boards.

Mine is a `Woodpecker 3.0.2 GRBL`:
[image]

It has a `UART1`-connector, which seems to be for connecting a controller/display/jogger with SD card slot:
https://aliexpress.com/item/GRBL-offline-control-board/32868562540.html
https://www.ebay.de/itm/3-Axis-GRBL-Control-Board-with-Offline-Hand-Controller-CNC-Laser-Milling-Machine/273169033288

RaulJYepez, 2018-05-12:
Hello, I just want to share with you that I was able to update my Woodpecker 3.0.2 from grbl 0.9 to 1.1.

I use Xloader, the hex file of grbl 1.1 and this is the configuration that I am using:

$$ < $0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=2000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=800.000
$111=800.000
$112=500.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000
ok

Note: If you are using GRBL 1.1 you need to use the last version of "Candle"

Regards

hberjanin, 2018-06-25:
Hello Raul, I have doubts about how to upgrade GRBL 0.9 to 1.1 using XLoader: 
Which connection did you used? The same described in this picture:
[image]

Is possible to use the USB conector?
As refence I have this board version:

[image]

Thanks in advance!

tgirard, 2018-07-14:
Ok, I read this whole thread and I was able to upload 1.1 to my GRBL .9 board that I just bought as part of a  CNC engraver. Here is my Step by step with version numbers
MS Windows 10
Arduino Version 1.8.5 (Standard install)
PuTTY Release 0.70

Steps:
CONNECT THE BOARD:
Plug your board into your computer. Make sure you know what Port number it uses. 
It should look something like this:
[image]

BACK UP YOUR SETTINGS:
Connect to the board through USB using PUTTY (You may have to play with the baud rate) 
[image]
You should see the GRBL Prompt
[image]
(Sorry, I didn't take a screenshot before I uploaded but you get the idea)
Type "$$" hit Enter, this saves your configuration 
The GRBL Board will spit out all of it's config and say ok
Close Putty

UPLOAD THE NEW FIRMWARE:
Download the GRBL 1.1 HEX file "grbl_v1.1f.20170801.hex" is the latest (July 2018)
I got mine here: https://github.com/gnea/grbl/releases
Place the .hex file in C:\
Find your avrdude.conf file, COPY it "ctrl-c" and paste it in C:\ 
Mine was here "C:\Program Files (x86)\Arduino\hardware\arduino\avr\bootloaders\gemma"
So when your done, you should see something like this in your C:\
[image]

Open a Command Prompt (Type "cmd" on the start menu)
Now, if everything is setup like I have it you should be able to use this command at the prompt 

avrdude -CC:\avrdude.conf -v -patmega328p -carduino -PCOM12 -b115200 -Uflash:w:C:\grbl_v1.1f.20170801.hex:i

Copy and paste this into the CMD and hit enter
It should go through a bunch of stuff and look something like this:
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin>avrdude -CC:\avrdude.conf -v -patmega328p -carduino -PCOM12 -b115200 -Uflash:w:C:\grbl_v1.1f.20170801.hex:i

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\avrdude.conf"

         Using Port                    : COM12
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "C:\grbl_v1.1f.20170801.hex"
avrdude: writing flash (30492 bytes):

Writing | ################################################## | 100% 5.17s

avrdude: 30492 bytes of flash written
avrdude: verifying flash memory against C:\grbl_v1.1f.20170801.hex:
avrdude: load data flash data from input file C:\grbl_v1.1f.20170801.hex:
avrdude: input file C:\grbl_v1.1f.20170801.hex contains 30492 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 4.01s

avrdude: verifying ...
avrdude: 30492 bytes of flash verified

avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

If everything went as planned you should be able to open Putty and see the new version in the prompt

I hope this helps 
-Timmy

Orval14, 2018-07-17:
Hello, to answer hberjanin (I have the same card and I just update by following the method of tgirard).
[https://user-images.githubusercontent.com/34790588/41859756-2a9e87aa-789d-11e8-87b9-6badf7ffa74d.png](url)
The connections on your photo are the good ones. I used an Arduino Uno flashed like arduinoIsp to flash the bootloader for adruino Nano / Atmega328p (for the Woodpecker) directly in the application Arduino 1.8.4 (version 1.8.5 does not work or bad ...) . - unplug the Uart1 before flashing the bootloader -
Once done, you must be able to disconnect everything (the 6 wires and the Uart1 plug too) and go through the Usb port of the card and follow the method described by Tgirard.
Personally, I kept the Uno as usb programmer (I broke the mini USB plug :o) and followed step by step the method described by Tgirard except for the speed (an AvrIsp programmer and a 19200 baud rate for me).
The command to launch in window "cmd":
avrdude -CC:\Users\orval\Documents\CNC\Tools\avrdude.conf -v -patmega328p -cavrisp -PCOM7 -b19200 -Uflash: w: C:\Users\orval\Documents\CNC\Tools\grbl_v1.1f. 20170801Wpr.hex i

PS: in the development application I had each time synchronization errors (transfer speed), 10 tests and error ... In Cmd window, it works with each transfer without error!

If this could help you or comply with your research, thank you all for your information transfer and your work exposed here.
Greetings

tgirard, 2018-07-17:
Correction:
In my post above, I said you should type "$$" to backup your settings. I found that it really doesn't back up the settings. In the end though I believe that this isn't such a bad thing as it requires you to read and better understand your new board and also understand what the 1.1 GRBL release adds in the way of accessible  functionality. My apologies for the misstatement. I'm adding a note above in the post. 

Unfortunately, I have been having all kinds of trouble getting Candle or grblControl to work on my setup. 
For some weird reason, Both Candle and grblControl will launch after a reboot "Sometimes" and work, but once I close either of the apps, I can't get them to launch again. If anyone knows what I'm doing wrong or could give some step by steps I'd sure appreciate it.

Orval14, 2018-07-17:
For the $$, in PuTTY, you can activate the backup of the answers in the .log file (in session - loggin - x Printable output with a log file name). This will always help to set up his "new" machine more easily ...

tgirard, 2018-07-22:
Hey Orval14, could you give a step by step on how to do that here? What are the exact cmds you typed in Putty?

blourido, 2018-07-30:
Hello:
I just get grbl 1.1f in my CNC Machine.

The WOODPECKER 3.1 GRBL is the control board my CNC have, this is an arduino uno chinesse clon, but they didn´t install the bootloader and you can´t upgrade from USB port.

Make sure your board is the same because each Woodpecker version is different.

[image]

The easy way to upgrade from grbl 0.9 for me is to use an arduino uno. You can program this arduino as a programer. 

From the ARDUINO IDE "Quemar Bootloader"
[image]

You must conect this arduino pins to Woodpecker Board (look at the pictures):
Aduino                          Woodpecker
pin 10               to                   Rst (orange wire)
pin 11               to                   MOSI (green wire) (crocodile to mosfet)
pin 12               to                   MISO (white wire) (directly to Atmel mega 328P pin)
pin 13               to                   d13 (SCK) (yelow wire)
Vcc (5V)            to                   5V (red wire)
GND                 to                    GND (brown wire)

[image]

[image]

Finaly you can upgrade grbl from USB with the arduino IDE

[image]

Orval14, 2018-08-05:
Good evening,
blourido - your description is consistent with the allusion of my point 3 already described and well commented in the previous posts. For information, you can connect the tab "MISO" on the Zen connector. To have the same card as you, it works very well like this and it's easier to wire.
[image]

jonhwiki - You can repeat the description made by -blourido- You can even, at worst, invest in an arduino Uno ($ 3) to recover your Woodpecker. If it does not work look at the semi-manual method of tgirard and used the UNO remained cable like IPS to put back the correct version (beware the 19200 speed for me with a 3.1 card - see above).

jonhwiki, 2018-08-05:
Some more detailed video, try the arduino load the grbl and it does not work.

Orval14, 2018-08-06:
It's a bit short "it does not work"
What does not work?
- the card on the CNC,
- flashing the bootloader (error message?),
- the loading of the Grbl version (error message?),
- the program itself (test with PuTTY for example?)
Where does your update procedure stop working with which error?

blourido, 2018-08-07:
Dear Orval14:

I look at your Woodpecker board but it´s not the same as mine.
WOODPECKER 3.1 GRBL is different than WOODPECKER CNC GRBL0.9
Is important to look if your board is exactly the same, because each board is different and the pin board is different.

Orval14, 2018-08-08:
Good evening blourido,

I have a Woodpecker 3.1 as in your photos with the UART plug at the bottom for the small keyboard / miniSD card. Sorry, I took the picture of a 0.9 version of a previous post to illustrate my point. I am not at home to be able to take a picture of my card 3.1.
Actually the MISO lug is the one you specify but without a real connection I never managed to finish the update (synchronization error). By plugging the MISO on the Zen plug (electrical continued with multimeter checked) and soldering the MOSI on the regulator leg, the transfer worked the first time.

For those who encounter some concern with this temporary contact on this very thin lug of the microprocessor (I like your protection between MISO and MOSI), the Zen plug seems more simple.

Photo of the seller's site of my card - screenprint "Woodpecker 3.1 GRBL" - purchase June 2018
[image]

Orval14, 2018-08-08:
Small complement

For those who have a Woodpecker 3.1 card and want to solder a connector to flash the bootloader, you can connect to the holes at the bottom of the card like this:

[image]

gojimmypi, 2018-10-14:
@blourido and @Orval14 thank you for posting this information on the Woodpecker 3.1 GRBL board. I've ordered my first-ever CNC (a 3018, specifically [this one](https://www.aliexpress.com/item/New-CNC-3018-Pro-GRBL-Control-Diy-mini-cnc-Machine-3-Axis-pcb-Milling-Machine/32887161965.html)) and I am expecting it this week! :) 

Would you happen to have a schematic for the Woodpecker 3.1 controller? I found this one, but it is unlabeled. I suspect it is for a slightly different board:

https://s3.amazonaws.com/linksprite/cnc/schematic+2.pdf

I suppose the most important thing is the [image above showing the bootloader pins](https://github.com/gnea/grbl/issues/336#issuecomment-411334501).

Thanks

ccantill, 2018-11-25:
FYI, I found out there is a much simpler way of re-flashing my Woodpecker 3.0 board with just the onboard CH340 serial. Apparently the bootloader is loaded on the chip, but the reset pin isn't hooked to the CH340. The key is to briefly ground the reset pin and run avrdude at the exact same time. I used the two middle pins of the bottom row of the UART connector for this. Timing is critical here and starting avrdude and releasing RST from GND has to happen pretty much at the exact same time.

RaulJYepez, 2018-11-28:
Good day @hberjanin .
 
1. You need download the .hex file of grbl 1.1 in the next link: https://github.com/gnea/grbl

[image]

2. Conect your board 3.0.2 with the normal cable and in Xloader you have to select the hex file, the device is Duemilanove/Nano(ATmega328) the correct COM port and baud rate 115200

[image]

3. Upload the file (this step takes some minutes, don't go insane)

4. Configure the board inside of the control software (Candle in my case)

Regards

bmalbusca, 2019-02-06:
> Hello:
> I just get grbl 1.1f in my CNC Machine.
>
> The WOODPECKER 3.1 GRBL is the control board my CNC have, this is an arduino uno chinesse clon, but they didn´t install the bootloader and you can´t [...]

Hi, I have an error when I'm trying to burn the bootloader of woodpecker : avrdude: stk500_disable(): unknown response=0x12 . Wich configurations  Arduino ISP have you used? I0m using the Arduino Nano.

Orval14, 2019-02-06:
Hello, normally Nano or Uno it must work all the same if the wiring is good.
To have a little more information on the transfer, you can modify the upload.verbose = true line in the preferences.txt file to the end of the file (see its location in file-> preferences or [ctrl] + [,]) .
To also put the odds on your side, make this update the controller card of the Woodpecker unconnected and with its own power supply. The USB cable is only for data transfer (current less than 500mA).
You can reset (on the Woodpecker) just before starting the transfer (actions as simultaneous as possible).
Good luck

peagerfw, 2019-04-18:
Hi guys, I have problem with my two board woodpecker 3.2 - 3.2A
All this boards are updated on 1.1f and thats it all with my "tuning" I do not have any next upgrade.
1)Board 3.2
- when I working, board is always disconnect from usb port. (after 10min ,25min or 1hour always diferent time)

2)Board 3.2A
-Stepper motor Y,X not work. (spindle motor is ok)
-used software GRBL candle
-picture for info
(12v led not lighting,SEMI is very hot)
=Bad 47uH?
[image]
[image]

I have read many informations, but I can not find solution of this problem. Help me please. Thank you
Peter

Edwinrj, 2019-08-11:
I discovered a solution, what happens is that the boards are configured with an old bootloader and the arduino IDE recognizes it as an arduino nano in the old bootloader option (I am using version 1.8.9).  

[image]

I uploaded the GRBL 1.1 with the arduino IDE, place these settings (my  board woodpecker is v3.3)
    $ 1 = 25
    $ 2 = 0
    $ 3 = 5
    $ 4 = 0
    $ 5 = 0
    $ 6 = 0
    $ 10 = 3
    $ 11 = 0.010
    $ 12 = 0.002
    $ 13 = 0
    $ 20 = 0
    $ 21 = 0
    $ 22 = 0
    $ 23 = 0
    $ 24 = 25,000
    $ 25 = 500,000
    $ 26 = 250
    $ 27 = 1,000
    $ 30 = 1000
    $ 31 = 0
    $ 32 = 0
    $ 100 = 800,000
    $ 101 = 800,000
    $ 102 = 800,000
    $ 110 = 800,000
    $ 111 = 800,000
    $ 112 = 500,000
    $ 120 = 10,000
    $ 121 = 10,000
    $ 122 = 10,000
    $ 130 = 200,000
    $ 131 = 200,000

 and the CNC worked

martinglass, 2019-08-17:
Here's the stock settings from my Woodpecker CNC Grbl 0.9j board. Acceleration rates are much higher (like 5000 for X & Y) compared to the Default Grbl ones:

$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=2 (dir port invert mask:00000010)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

pxp888, 2019-08-21:
I have the woodpecker CNC board version 3.3.  Are the pins for ISP programming the same as 3.2?

[10 more replies]