gnea/grbl #337: Incorrect sanity check of spindle configuration values in grbl.h

nicolas-lang, 2017-12-30:
In grbl.h there are sanity checks for the #define configuration.
In row 71 is a error check for the spindle configuration: `#if defined(USE_SPINDLE_DIR_AS_ENABLE_PIN) && !defined(VARIABLE_SPINDLE)` and config.h contains a corresponding comment `// NOTE: This configure option only works with VARIABLE_SPINDLE enabled and a 328p processor (Uno).`

This check is in my opinion neither correct nor usefull.

The usecase which this collides with, is an simple "on/off" spindle without direction or PWM.
When configuring this usecase my assumption would be to #define `USE_SPINDLE_DIR_AS_ENABLE_PIN` (with both outputs configured to be on the same port/pin) in order to disable the output of an direction signal, and remove `#define VARIABLE_SPINDLE` to disable PWM.

In my opinion there is no code in spindle_control.c that would break by this configuration.

In summary i would consider this check unnecessary/incorrect and would suggest to revisit it's substantiation with the target of to removing it.

chamnit, 2017-12-30:
@nicolas-lang : I think you're misunderstanding what options there are:

The default with spindle direction (D13) and spindle PWM(D11).
The SPINDLE_DIR_AS_ENABLE_PIN with spindle enable (D13) and spindle PWM (D11).
VARIABLE_SPINDLE disabled with spindle direction (D13) and spindle enable (**D12**) to support the older v0.8 pin configuration. 

You need the two checks to ensure you are using the newer v0.9+ pin configuration. Otherwise, you are taking time to argue semantics.

closed 2017-12-30 by chamnit

nicolas-lang, 2017-12-30:
@chamnit 
As far as i understand your argumentation, it is based on the assumption of a static pin configuration/CPU map for each specific GRBL Version.
My assumption was, that a custom wiring/configuration is possible, and only requires the user to modify cpu_map.h and config.h based on the existing hardware/ desired behaviour. 

Thank you for the clarification.