gnea/grbl #338: Understanding control state logic

dastultz, 2017-12-30:
Hello, I'm running GRBL 1.1f. I have only one "control" wired up and that is `CONTROL_SAFETY_DOOR_BIT`. The switch is Normally Closed so I also
`#define INVERT_CONTROL_PIN_MASK ((1<<CONTROL_SAFETY_DOOR_BIT))`

Looking at the code [here](https://github.com/gnea/grbl/blob/master/grbl/system.c#L47), we enter the block if at least one pin is raised. Then we ask a number of `bit_isfalse` questions. This seems odd to me. If something is `true` we enter the block and try to find something that is `false`. If my door is ajar, the result of `bit_isfalse(pin,(1<<CONTROL_SAFETY_DOOR_BIT))` is `true`. I'm not sure it matters that I've got just the one control, but when the `CONTROL_SAFETY_DOOR_BIT` is not set, the test on `pin` is also false and it never enters the block.

I have to comment out the block `if (pin) {` in order to get my door switch working.

Have I done something odd? Unexpected? How does this work for anybody at all? I expect `CONTROL_SAFETY_DOOR_BIT` to be `true` when there is a problem, so the `if (pin) {` seems proper, but then I would expect a test on `bit_istrue` on the `CONTROL_SAFETY_DOOR_BIT`.

Am I making any sense?

Thanks.

/Daryl

dastultz, 2017-12-30:
I should add that I also have to comment out `CONTROL_RESET_BIT`, `CONTROL_FEED_HOLD_BIT`, and `CONTROL_CYCLE_START_BIT` inside this `if` block to get the behavior I want, if I don't I get all kinds of wackiness when I trigger the door (resulting in an Alarm state).

langwadt, 2017-12-31:
I normally default to assuming grbl is correct since so many successfully  use it, but I agree it looks strange and seems to only work because of some quirk. and it is inverted compared to the old grbl code

shooter64738, 2017-12-31:
Did you also enable the control_mask inversion? @ line 185 // #define INVERT_CONTROL_PIN_MASK CONTROL_MASK // Default disabled. Uncomment to disable

The pin variable is masked with CONTROL_MASK and I believe thats why the 'pin' value is remaining zero and not entering the if block. Essentially that if block is checking to see if 'pin' is more than zero.

dastultz, 2017-12-31:
@shooter64738 please see the first post, I included the reference to `INVERT_CONTROL_PIN_MASK` I am using, did I not do it right?

shooter64738, 2017-12-31:
Yea I saw that, but there are 2 masks next to each other. I wasn't for certain which one you were referring too. 
After reviewing it a bit more, pin is getting its bits set, then if INVERT_CONTROL_PIN_MASK is defined, we invert the pin bits on line 47... So pin should be some value above zero and enter the if block. But then.. we check to see if the door bit is false/0.. and it cannot be false and true at the same time.. 

Im leaning towards, if inversion is on and it switches a 0 to a 1 to get into the if block, it should be checking for the bit to be true. I see your confusion. :)

dastultz, 2017-12-31:
Looking at the history of this function, Sonny Jeon renamed and reorganized some things and also inverted all the tests. Comment says "Moved (and tweaked) the invert control pin mask define and placed into config.h, rather than in the cpu_map.h file. Makes more sense there." This change was made on 3/30/16. I don't get it...

dastultz, 2017-12-31:
As I mentioned, my door switch is NC which, if I understand anything, is a "1" or "true" in computer parlance. That's why I inverted it, thinking I wanted it "true" if the door is open. I removed the inversion and reverted Sonny's change and it now behaves the way I want it to. The code in question now looks like this for me:

```  if (pin) {
    #ifdef ENABLE_SAFETY_DOOR_INPUT_PIN
      if (bit_istrue(pin,(1<<CONTROL_SAFETY_DOOR_BIT))) { control_state |= CONTROL_PIN_INDEX_SAFETY_DOOR; }
    #endif
    if (bit_istrue(pin,(1<<CONTROL_RESET_BIT))) { control_state |= CONTROL_PIN_INDEX_RESET; }
    if (bit_istrue(pin,(1<<CONTROL_FEED_HOLD_BIT))) { control_state |= CONTROL_PIN_INDEX_FEED_HOLD; }
    if (bit_istrue(pin,(1<<CONTROL_CYCLE_START_BIT))) { control_state |= CONTROL_PIN_INDEX_CYCLE_START; }
  }
```
Still confused as to why I should have to make such a change (and why Sonny made his change).

chamnit, 2017-12-31:
Just remember that the `pin` variable reads the state of the port pins. 1 is high, 0 is low. Just keep track of that and it'll be easy to discern the few lines of code there.

closed 2017-12-31 by chamnit

dastultz, 2018-01-01:
It's not actually important for me to understand why it works, although I am a programmer by trade so I'll probably lose sleep over it...

As I said, the only CONTROL I am interested in is the door, so I set this in `cpu_map.h`:

```#define CONTROL_MASK      ((1<<CONTROL_SAFETY_DOOR_BIT))```

Since I wanted to invert it, I set this in `config.h`:

```#define INVERT_CONTROL_PIN_MASK  ((1<<CONTROL_SAFETY_DOOR_BIT))```

This doesn't work. I got the idea to invert all of the control pins, I replaced the above with this:

```#define INVERT_CONTROL_PIN_MASK ((1<<CONTROL_RESET_BIT)|(1<<CONTROL_FEED_HOLD_BIT)|(1<<CONTROL_CYCLE_START_BIT)|(1<<CONTROL_SAFETY_DOOR_BIT))```

And this works as desired, though certainly not intuitive. I don't get it, but whatever, it's working. Thanks.

chamnit, 2018-01-01:
@dastultz : Wait. If it's not important for you to understand why it works, then what was the point of tracking down a minor code change in a 3/30/2016 pull request and posting this question: "**chamnit can you explain the reasoning for the logic inversion here (bit_istrue to bit_isfalse)?**"

To make it explicitly clear once again, the ```bit_isfalse()``` change was made for code conciseness. More specifically, it removes the need to invert the entire pin variable inside a time-critical pin change interrupt and saves a few bytes of flash (literally 100 bytes left on the 328p for certain critical configurations). 

In addition, keep in mind that Grbl also has to support inverted pin states, where a pin can be inverted to normally low and a pin high is triggered, rather than normally high and pin low as triggered.  In this case, the ```bit_false()``` can be now interpreted as "right" according to your definition. In fact, the ```bit_isfalse()``` check does not explicitly denote either triggered or not triggered. It's ambiguous.  It's not asking if the pin is triggered. It just checks if the bit of the pin variable is false.

Honestly, I really don't see what the big fuss is about over a few lines of compact code. Code like this is common in scientific research (my background). You are constantly trying to balance optimizing for performance with readability. To me, this code is quite clear if you take the time to go through it with the understanding that the pin variable fetches the pin states on the port and you may have different trigger states.

dastultz, 2018-01-01:
I asked for an explanation so that it might help me understand how to configure my setup. I should have asked for more direct help, sorry about that. GRBL is a great product, I'm amazed at how well it does what it does. Thank you for it. I'm not a great C programmer nor a great microcontroller programmer. But I am a good programmer. Looking simply at the block in question outside of its context this is the way I see it (simplified code):

```
 if (pin) {
      if (bit_istrue(bit_1)) { do_a_thing }
      if (bit_istrue(bit_2)) { do_a_different_thing }
      if (bit_istrue(bit_3)) { do_yet_another_thing }
  }
```

I see `if (pin)` as an optimization. We are interested in the true state of a number of pins. If none of the pins are true, there is no need to test each pin. The code was changed to this:

```
 if (pin) {
      if (bit_isfalse(bit_1)) { do_a_thing }
      if (bit_isfalse(bit_2)) { do_a_different_thing }
      if (bit_isfalse(bit_3)) { do_yet_another_thing }
  }
```
Now we are interested in the false state of the pins. Why should we require at least one pin to be true to enter the block? If all the pins are false we should take all of the actions, but we would not take any because of `if (pin)`. So I contend that `if (pin)` is no longer a logical optimization. I was forced to configure pins I was not interested in order to get it to work.

Sorry if I offended.