gnea/grbl #339: CoreXY homing issue

bajticzek, 2017-12-31:
Hello, I built a CoreXY machine with Arduino Mega and RAMPS 1.4. I am using grbl-Mega fork v1.1f but since I think the Mega fork has only minor changes, this issue could be answered there too.

Basically I have a problem with homing - only one motor moves at a time. That would be fine with Cartesian machine, but it is a problem with CoreXY one. One thing to mention - jogging works fine, moving in either X or Y moves both motors at a time, so the motion is in the direction I want. But that is not the case with homing.

I altered these lines in config.h:
- uncommented #define COREXY
- added #define INVERT_MIN_LIMIT_PIN_MASK ((1<<X_AXIS) | (1<<Y_AXIS)) - limit switches now work properly
- uncommented #define HOMING_CYCLE_0 (1<<X_AXIS) and #define HOMING_CYCLE_1 (1<<Y_AXIS)
- uncommented #define HOMING_FORCE_SET_ORIGIN - I am used to 3D printers so for starters I would like to use "positive workspace"

I changed these eeprom constants via commands:
- $0 = 3
- $1 = 100

Any ideas how to make homing work properly?

chamnit, 2017-12-31:
@bajticzek : This issue would be more appropriate in the Mega repo. It looks like a bug related to the experimental RAMPS support code, which was written by a helpful user. Not the main line default code. It's likely that the RAMPS code doesn't support homing on CoreXY at this time, but it might be an easy fix.

closed 2017-12-31 by chamnit

chamnit, 2017-12-31:
Looked at the code. Didn't find anything obviously unusual. It should be working. Please re-post this issue at the Mega repo. Include your $$ settings, $I build info output, and all alterations to the default source code (config defines, etc).

bajticzek, 2017-12-31:
Thank you, will do.