gnea/grbl #340: Possible hard limits issue

jasoroony, 2018-01-01:
In the LIMIT_INT_vect function you call mc_reset() and then set EXEC_ALARM_HARD_LIMIT - which gets picked up by a protocol loop that looks to want to wait for the user to do something - but it doesn't wait because of the flag set in mc_reset (sys_rt_exec_state = EXEC_RESET).

protocol_exec_rt_system runs the loop once and sees it's time to leave because of the reset flag.

Perhaps saving and restoring sys_rt_exec_state around the call to mc_reset() in the hard limits interrupt would be better?

chamnit, 2018-01-01:
@jasoroony : Are you sure about this? Did you check this on the Arduino or just looking at the code? 

There is a clear EXEC_RESET flag prior to entering the wait state in the critical alarm code. It should wait there until the user has sent a reset command.

jasoroony, 2018-01-01:
So there is, that makes sense now - this isn't on an Arduino, I'm trying to get hard limits to work on an LPC1768 board that (like most) don't put the limit pins on an interrupt port.  My polling routine (that re-purposes the limit interrupt routine) is what's triggering the loop exit.  I call it inside that loop along with other things that need polling.

Thanks!

closed 2018-01-01 by jasoroony

jasoroony, 2018-01-01:
Rhetorical question to think about - doesn't another limit interrupt then reset the machine on a real Arduino?

jasoroony, 2018-01-01:
This is strange - I see the code that prevents a second alarm being set in the interrupt while already alarmed.  It's now not making much sense at all - I'll have to investigate some more.  But I have not changed much of the original code (and doing what I suggested in the first post did work for me).

jasoroony, 2018-01-01:
Ah, never mind.. it's my first polling call being in the wrong place.  After entering protocol_exec_rt_system(), the [only] local variable gets assigned to sys_rt_exec_alarm - if my first polling call is done immediately after that (where it was being called) then the issue arises.  Moving it to be immediately before it fixes everything.

Sorry for the inconvenience.  (Oh, and I know polling isn't a great solution - but I think it's better than none...  Actually it's not half bad, just running a test on this 100MHz chip now, it stops very quickly.)

chamnit, 2018-01-01:
It’s not clear what you are polling and when, but it sounds like you are not using an interrupt for the hard limit. If polling is fast enough, it should work ok but depends on it being polled consistently and on time.

jasoroony, 2018-01-01:
I should re-open this issue, but I'll instead leave it with you to decide..  After some needed sleep, I do now believe my use-case above has shown exactly when the limit interrupt on a real Arduino is unsafe and causes the problem I first described.  It shows the very brief period (a handful of assembler instructions) in protocol_exec_rt_system() where between the first statement, the initial alarm assign to the local, and the next state assign to that local (after the alarm test if statement) where the interrupt (that by definition can happen there, however unlikely) will cause an unwanted reset.

I'm glad I put "possible" in the issue title, because while it is improbable - I'm pretty sure it can still happen in theory.

chamnit, 2018-01-01:
I suppose it’s possible that a hard limit occurring right after the alarm check and before the reset check could lead to it missing the alarm. Fortunately, the chance of this is very very small and the machine will have still come to a stop due to the mc reset call. 

There are a few ways to fix this like making these checks atomic. But I tend to avoid making any major changes to the code on any master release like v1.1. I also consider Grbl v1.1 the end of the line. I’m in the middle of a major refactoring for the upcoming ARM release, which includes the state machine overhaul. I’ll make sure that this is addressed there.

jasoroony, 2018-01-02:
Hmm.  Fair enough.  With it not affecting me, I'm indifferent to fixes, however you will probably not be surprised that a lot of software at end-of-life gets used for 20 or so more years.  I'm also sure my real Arduino machine (I've got one!) has reset unexpectedly around minimum values (that from today I won't lose sleep over thinking about it).  Trust me though to find the perfect place not to put some desired change in - in some random project - where it should work, but doesn't.