gnea/grbl #341: Startup message sends no "ok"

pixartist, 2018-01-01:
This is really a pain, how am I supposed to correctly handle the startup when it does not send an "ok" message when done?

chamnit, 2018-01-01:
Can you be more specific on what’s happening?

swarfer, 2018-01-02:
I dont see a problem.   
on startup you get the startup message.
in reply to any commands you get the 'ok'.
so you can only expect an 'ok' if you sent a command (excepting a restart command)

pixartist, 2018-01-02:
It is nowhere defined how long the startup message is, and there is no ok at the end. So how am I supposed to know the startup has finished ? For example when hard limits are enabled, the startup message will contain two lines. If not it will only contain one. So where can I exit the thread that is waiting for the end of the start message and go to the send/recv thread ?

chamnit, 2018-01-02:
Please read the interface wiki again. This is not a difficult problem to solve, as you can pull ideas from the numerous other Grbl streamers that are open source. This is just a matter of checking the message type and dealing with it appropriately.

closed 2018-01-02 by chamnit

cheton, 2018-01-03:
Hi @pixartist 

You can check out the regular expression below to learn how to parse Grbl startup message:
https://github.com/cncjs/cncjs/blob/master/src/app/controllers/Grbl/Grbl.js#L564.

It can handle customized Grbl firmware (e.g. gCarvin) as well.

pixartist, 2018-01-03:
I already do that, I did not realize grbl's feedback message came up whenever grbl felt like it. I thought the alarm was part of the boot message.

chamnit, 2018-01-03:
@pixartist : Yes. Grbl will send you messages when it needs to. It's called real-time or push messages in the wiki. Otherwise, you'll have no idea if something happened until the next send-response cycle of the streaming protocol. You just need to filter out the push message. It's not hard to do.