gnea/grbl #342: GRBL response formats are lack of consistency

Dani88, 2018-01-02:
Disclaimer: it's not a big deal and working with GRBL is awesome in any other way, but it still bugs me - and bugs me very much, exactly because GRBL is so consistent in any other way.

We have plenty of different type of status/settings/configs/options/parameters, all are unique and definitely required, but the way you can identify and parse them are very different, and sometimes illogical with lots of exceptions. 

Let me summarize the most important ones:

1. StatusReport
_Request-Response_: One realtime command => one line response without 'ok' (perfect!)
_Detection_: Easy. First char is `<`.
_Parsing_: Two levels: split by pipes for the key-value pairs, and split the pairs by `:`. (However that's not true, see the concerns)
_Concerns_: 
- The status itself (as an enum) is an excpetion. Why is there no just a single char of key for this value? Like S:Alarm, or S:Idle. Yes, it's extra memory, but you can get rid of the string status labels, use simple integers (see error and alarm codes, or setting keys). 
- But there is a second level excpetion within the status, because it has sometimes (but not always) a sub-state (Hold:0). Why they are not a specific kind of states? Or why don't we have sub-states (with default 0) for the other states in the name of consistency? If all the states are just ints, you can have plenty of states/substates.
- WCO is a tricky one, because it's kinda an exception here, but WCO is redundant data anyway, because you can get the current coordinate system from the ParserState ($G) and get the offset for that coosys from the GcodeParameters ($#). But okay, it's nice to have, and because of the key-value pair format, you can easily parse the response when the length is dynamic.
- BufferState (Bf), Speed (FS), OverrideValues (Ov): all are exceptions because you have more values for one key. Even worse that they are in specific order. Sometimes we have key-value pairs and dynamic length of response, but sometimes not. In this case within one thing (status report) we see the two different approaches. Of course I can see the logic behind them, GRBL has very limited amount of memory, and extra keys would make it worse, but let's be honest: use integeres for the keys (instead of 2 chars key) and there you go. You can easily identify all the data with an exact key, and eats less memory.

_Idea_: <key_integer:single_value | >, for eg: <0:2|1:1,2,3|2:35| ... > Strict and consistent format, less memory consumption, interface writers can use enums for the keys for more readable codes. Yes, not human friendly...

2. Settings
_Request-Response_: One stream command => multiple lines response with 'ok' (worst case scenario)
_Detection_: Easy for one line, very hard and risky for the full setting list.
_Parsing_: The same as detection, easy for one line, but hard if you want the whole picture.
_Concerns_: 
- Here we have a very consistent key-value pair list, with integer keys (as I advised for the status report above), but we have a new separator: '='. That's the only case when the key-value separator is '='. Yeah, it's not a big deal, like this whole thread, but still, worth to mention. These are just my small concerns. It's just extra complexity without any benefit.
- This is my biggest issue: The whole one line request, multiple lines response stuff is killing me. Yes, we have an 'ok' at the end of the list, just like we have one when GRBL accepts the $$ request,  but it means, you cannot have something like a generic ResponseProcessor<T> which accepts a response and handle it. In this very case the read and write parts are tightly coupled, a CQRS solution is could be pretty tricky. It looks like, I have two option at this point: ignore the OK message at the end of the list and go with the easily identifiable lines which start with $<number>, in which case I have no idea when all the settings are known for me, or option#2: write a mini state machine only for the setting printout, which knows about the starting and ending ok. That would be cool, if it would be useful anywhere elsewhere, but it's not. It's just the big extra complexity for one time. Also there is the possibility that something terrible happens meanwhile I'm listing the settings, for eg a limit switch triggered, which will send an immediate push message in the middle of my state machine.

_Idea_:  
- First and foremost: get rid of the multi line solution. There should be only one line response for each request. Use the standard ':' key-value separator and the pipe between them. Yes, human readability again, but I'm not sure how much should it matter when it's about interfacing, and since line endings play a big role in our communication, it should be used very carefully, not like this. $0:1|1:123.456|2:3 ... Looks ugly? Yes, just like the status report, but still, UI writers can display it whatever fancy format they wish to. Or make a configuration in config.h for this format if readability is such a concern.

But it getting worse:

3. GCode Parameters:
_Request-Response_: One stream command => multiple lines response with 'ok' (worst case scenario)
_Detection_: Nightmare. There's no easy way to identify a parameter response only if you exactly know what you expect (again: read and write tightly coupled, or the responses are hard coded)
_Parsing_: Inconsistent. The key-value separator is ':' but there's exactly one exception without any good reason (PRB has two values, with the very same separator)
_Concerns_: 
- The one request => multi line response is the worse here since you can't even identify a single line easily. The first character is commonly used elsewhere (all the feedback messages start with '['). The only solution is a hardcoded list, which contains all the optional keys (G54, G28 G92, TLO, PRB, ...). That is the only case, when you cannot identify a line at a glance. No special starter character, no static key. Nothing.  If we would have a special character to identify a line as parameter response it would be at least the same as the Settings. But no. It's a 3rd case, without a single good reason.
- The PRB parameter makes it even worse, because we have 2 values for one key but with the very same separator. 

_Idea_:  Anything else. That's just stupid. All keys should have one and only one value. All the lines must be easily identifiable as a parameter response. And the multi line situation makes it even worse. Like always, the solution is the very same as always (exactly that's the point of the consistency):  One request => one line response, with key-value pairs (where the keys are integers). ':' as key-value separator, and pipe as pair separator. 

4. GCode Parser State:
_Request-Response_: One stream command => one line response with multiple data.
_Detection_: Could be much better, since the first char is not dominant here, you need to read the start of the line to identify the line as parser state. It's just impossible to use something like this:  https://i.imgur.com/WmcjIRR.png You have to write sub-processors without any reason for feedbackMessages.
_Parsing_: Not fun, but not bad. I like the way it works tho. The T/S/F parameters are exceptions, but I have no better idea for it. It makes sense, that T/S/F is part of the parser state and they are numbers, so yeah. I can live with it. A separate parser state response only for these 3 parameter would be too much I guess?
_Concerns_: 
- First char is not dominant (feedback message, but it's a special kind of feedback message). Why do we have to reuse feedback message so many times? Parser state is a very important information, and the UI use it heavily. It deserves a dedicated starter char (like all the responses). I mean, I can live without unique first chars, but it looks like we have a few of them, why don't we have for all the responses? Consistency again.
- Also the T/S/F parameters are exception, but I have no better idea, so let's skip this one.

_Idea_:  Replace [GC: to a unique char and I'm a happy man (who cares, I know)

I have many other concerns/ideas like these, but I it's already too long. I'd be happy to have a discussion about these.

And yes, my english is awful, and I probably made hundreds of errors. I'm working on it, please excuse me on this one.

winder, 2018-01-02:
I didn't notice anything awful about your English. The format you used along with constructive suggestions was very easy to follow.

> 1. StatusReport
WCO is a tricky one, because it's kinda an exception here, but WCO is redundant data anyway, because you can get the current coordinate system from the ParserState ($G) and get the offset for that coosys from the GcodeParameters ($#).

I sort of agree with you here, but the WCO data isn't strictly redundant for one reason: it is included in the next status message after it changes. You're much more likely to send a status request than a `$#` request.

> 2. Settings 
First and foremost: get rid of the multi line solution

One of the first troubleshooting steps we ask is "send me the output to '$$'", so this suggestion would make support more difficult. A lot of GUIs also augment the single-line responses with the [CSV translation](https://github.com/gnea/grbl/blob/master/doc/csv/setting_codes_en_US.csv) which makes this even more helpful during support.

The suggestion of using `$0:1|1:123.456|2:3` is an interesting one though, I do like the consistency but am not sure that outweighs the usability cost.

> 3. GCode Parameters:
The only solution is a hardcoded list, which contains all the optional keys (G54, G28 G92, TLO, PRB, ...)

You kind of answered your own question on this one. The special starter pattern is `[G54`, etc. They are called out as such [in the documentation](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#message-summary).

The `[PRB` comment is a little alarming though, I didn't realize that was in there and it might be a bug in some senders if the user were to sending `$#` during certain operations.

> 4. GCode Parser State:

Seems this is a one-character concern again. I think having all the states in one line is a historic CNC convention which might be why GRBL is done this way.

The themes here seem to be multi-line responses and that you don't like multi-character patterns. But in both of those cases, you're claiming that this prevents you from having a stateless interpreter for the GRBL responses. That isn't true (except perhaps the `[PRB:` case). It is true that if you want to implement your parser with a <char, callback> dictionary, so I would suggest finding an alternate design.

You can find many examples on GitHub if you're curious, although I'm sure you don't need it based on the amount of detail in your critique. They might not strictly match one of Martin Fowlers design patterns, but many aren't especially complicated either. Here is mine, written 5 years ago back when I didn't know what CQRS was. Not only does it parse GRBL responses one line at a time but it can process any GRBL message from 0.7 to 1.1. I even named it similar to the generic `ResponseProcessor` you wanted: [rawResponseHandler](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblController.java#L142). For completeness the status parsing code as well, in my code I think this is the most complicated portion (for GRBL 1.1): [getStatusFromStatusString](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/GrblUtils.java#L302).

My biggest concern is that the units aren't explicitly included in the status messages, even something like the intermittent `WCO` would do. They are implicitly included by counting the number of decimal places though, but I don't like having to depend on that undocumented behavior.

Dani88, 2018-01-03:
First of all, I'm very happy that a contributor picked up my thread and gave me a straight answer. 

> [...] this prevents you from having a stateless interpreter for the GRBL responses. That isn't true (except perhaps the [PRB: case). It is true that if you want to implement your parser with a <char, callback> dictionary, so I would suggest finding an alternate design.

You just summarized my main goal in one sentence perfectly: "having a stateless interpreter for the GRBL responses". Yes, it is possible right now, I just did one, I just don't like it, let me show you the way I parse the answers:

1. High level identification: In most cases the first char is more than enough to identify the answer, which is the cleanest, fastest solution if you are using a dictionary: https://i.imgur.com/WmcjIRR.png

2. Feedback level: For feedbacks I'll need to get the key of the message first. Very similar to the 1. level identification, but still different: https://i.imgur.com/TqOlMDn.png

3. Gcode parameter level: There's no fancy/fast way to do it, just a pre-compiled list of the options: https://i.imgur.com/Cwq6Qw8.png 

It's deferentially doable, but you will need 3 different approaches, and I cannot see the point/benefit of the extra complexity which hits back hard when we are talking about maintainability, but that's only the tip of the iceberg.

Let's suppose 2+1 things:

1.  We have a fully implemented GUI for GRBL. Every single little feature is supported in at least one way. One of the most important requirement for this is an intuitive/user-friendly visualization. For that you'll need all the information what GRBL can provide. All the offsets, parameters, small settings (even from the config.h).

2. And a new version of GRBL is incoming. Some of the code breaks, maybe not everything is back compatible, it's not an issue, because you already expected that, so you wrote your GUI around easily modifiable configurations. A new setting kicks-in? Just add a new line to your Settings.xml/json with the details and use your (let's say) T4 template to generate the new code. The same true for system commands, or override feature, or even for status report and gcode parameters. It's a child play to write your code in such a way that your GUI can run with all the existing GRBL versions. They are just small extra information in the configs. TLO Parameter is a new one? Let's add a version=">1.1".

3. Stateless interpreter 

Now, we need all the information as soon as possible. The visualization needs to know about the size of the CNC, so first of all you have to query the settings on connecting (we don't trust in cached data on first launch). But it would be nice to show for the user the saved coords (for eg G28). So you'll need the GCodeParameters as well. The current modal state is another very important bit of information. Get that one too. 

Where do you know, that - let say - the settings are loaded? The bedrock of the communication is the \r\n, but you cannot use it. One line is not the whole picture just a fragment of the full requested data. 'Ok' is one kind of response, and $132=0 is another. I can publish an event which says, hey, the ZMaxTravel information is ready to use, but you cannot be sure that the same true for the XMaxTravel data. I really didn't want to touch the 'ok' response in the SettingPrintoutParser, so I had to add a middle layer which catches all the published events from SettingParser and monitors them. It's a check list basically which makes sure that you have a fully loaded setting model, and now your GUI can decide which feature to use and which not. One request => many answers => aggregator catch them all and publishes only one event: here's your settings. 

The only way your parser can be stateless, if you add a stateful services on the top of the communication layer - an aggregator if you wish. At the end of the day, you will have to know the previous responses to be sure a request is served. You can push the problem to higher level, and say that the parser itself is stateless, but that's just one side of the coin. The problem is still the very same, but you hide it.

I can't highlight enough: we all know, that it's very easy to write even a very complex GUI for GRBL, and if a small component cannot be written in a stateless way: who cares? But on the other hand: that's looks like a design flaw which easily fixable and backward compatibility with the existing GUIs is just question of a flag in the config.h.

chamnit, 2018-01-03:
@Dani88 : One word of advice. If you write a long post, don't expect an immediate reply. Try to keep your issues short. Otherwise it becomes TL;DR. I didn't have time until this morning to sit down and go through your points. 

- Status report state: It states in the interface wiki that the state value is always first. So you don't need an identifier. Messages take a lot of time and bandwidth, especially when streaming is heavy and status reports are requested at a high rates. The identifier is redundant and was removed to cut down on the message length.

- Substates: This is semantics. You can argue that having substates is more complicated. For example, 'Hold:1' would be 'Holding', and 'Hold:0' would be 'Hold Complete'. Again this was done to try to minimize message length without overly complicating things.

- WCO: This is absolutely required. A '$#' message can only be called when Grbl is IDLE. This is due to how EEPROM is accessed. Grbl doesn't have the memory space to retain everything, so it has to fetch it. Also, WCO can change dynamically during a job. For example, tool length offsets will do this. To reduce redundancy, WCO is intermittent, sent automatically when something changes, and can be set refresh up to every 255 status reports. This was also done in this way to help clue in users that there are machine and work coordinates. It would not be obvious if WCO wasn't persistently present.

- 'Bf:', 'FS', 'Ov' : Again, semantics. I'm sorry that you are annoyed by this, but this is easily solvable. I good parser shouldn't care what the order is or the label.  Integer identifiers are not used because they are not human-readable. I prefer simple character identifiers so I can quickly see what is going on. When you are looking at a stream of data in realtime from a terminal, it's a lot easier to discern the status message if they are in a fixed order. If they are always out of order, you'd constantly be hunting.

As for the second half of your post regarding 'ideas', I'm having a hard time following this. If you mean that you want to improve the comm protocol to include sending more than one line, I'm already working on this. It's hard to make it as simple as the current protocol, but I think it'll work.

I'll be first admit that Grbl's comm protocol isn't perfect, but it works once you've dealt with its quirks. Having a lot of dependencies makes it very difficult to alter things significantly. You'd break all existing GUIs and anger a lot of people. The switch from v0.9 to v1.1 style protocol was done very carefully. I got the input of all of the major GUI devs and OEMs to see if they had any issues with the new v1.1 protocol. It took a long time to get things sorted out. Yet, it's not perfect, but it's good enough. 

With the new ARM version, I have the flash and memory to afford trying a few things and still support the v1.1 protocol. When things get published, I'd be happy to discuss ways to improve it then.

Dani88, 2018-01-03:
Thanks. I really didn't expect a quick answer for this, I think it should be more like a discussion than a bug report or help request. I would appreciate your answer even if it would takes two weeks. But you just replied to it within a day. Thank you.

Based on you answer I have a feeling that it looks like I think GRBL has serious issues with communication which must be fixed. _That's not true_. There's no bug or error, the communication is already working in many different GUIs, the documentation is complete, without holes.

You mentioned semantics in your answer multiple times, and exactly that's what I was talking about. Yes, status is the first field in the status report response, just as the documentation states it, that's all clear. The documentation is accurate, there's no undocumented behaviors and so on. 
My point was that I see many inconsistency exactly in the semantics. Some of them can be easily solved some of them are not. That's a bit foggy if you word like this, that's why I created a simple list with examples and "ideas" which were meant to be constructive, and not some kind of claim. Sorry if it was equivocal.

Status report is not the perfect example but your answer is based on the status report itself, so I'll carry on with this one, even if status report is a bit too specific and my whole post is about the consistency of semantics around all the responses not just within one.

First of all: thanks for the WCO clarification. Totally make sense, it's not a duplicated data. 
I'll try highlight my concern based on your anwser.

> This is semantics
Exactly.

> state value is always **first**
Okay, so we have an order, like State goes first. Or within the Ov/Fs/Bf fields we have an order. That's far from the whole picture since we have tons of fields with label/key and without exact order.

> good parser shouldn't care **what the order is or the label**
At the same time, the parser should not care about the order. Ok, that's getting harder. Sometimes it should care sometimes it shouldn't. 
A parser must care with one of them, and only with one of them. We have an order or we have keys. Yes, we can have both, like we do have now, but that's just extra complexity for no reason. It works? Yes. It could be more elegant with only one approach? I think so, yes.

> identifier is redundant and was removed to cut down on the message length
But still, we have: {MPos, WPos, WCO, Pn, Ov, ... } with labels/keys. With a very good reason: it's readable even on a terminal. Which perfectly make sense and also explains why does status has no key the same way: because the status itself is readable and easily identifiable.

Probably it gives you an idea about my point of view. I understand yours with the easily readable responses, so let me approach it from this way:

It would be nice to have a machine friendly way to communicate with GRBL. And before the "good parser" strikes again: a good parser alone is nothing and easily achievable. You can send me responses in whatever format you wish to, and while it's consistent around all the messages I'll be happy with it. Let's go with human readable string labels, or with a struct layouts, or with integer keys. Does not matter. But let the semantics be strict and consistent, without L'art pour L'art exception. 

By the way, as I stated above, my primary concern is still the multi lines response, which can be explained with human readability again, in which case I can just repeat myself: a machine friendly solution would be nice on the ARM version for example, in the far future of course.

Since all the responses are human friendly now, would it be possible to add alternate response formats for machine communication where we don't have to hardcode edges cases and exceptions?

For eg keep $$ as the human friendly settings print out request, but add a whatever ascii code request for getting the settings  within one line and within a very solid format. The GUI doesn't need fancy keys or labels, or status strings, they just make it worse. With this solution a GUI could be more robust and the communication could be even faster since the messages would be smaller. One rule is better then 42 rules with 35 exceptions, if we are talking about machine communication.

chamnit, 2018-01-03:
@Dani88 : Your concerns are heard and are already being worked on. Grbl on the 328p can only afford the space for one protocol. That protocol has to be human-readable. It also had to remain close in compatibility to prior GUIs to prevent complete re-writes. A lot of the issues you state go all the way back to stuff I first installed in Grbl v0.7. Those decisions and mistakes have been grandfathered in to today's v1.1. Startup message and multiple lines are examples of those.

Like I mentioned before, I intend to experiment with other comm protocols with the ARM release, since I have plenty of space to do so. Your multiple lines concern can be easily eliminated. Other things like everything fitting into nice compartmental boxes can be done, but I've found that not everything fits into a single scheme efficiently. The biggest limiting factor is the minimization of message length and overhead. If you try to put everything into JSON, the messages quickly become huge, full of redundant characters. If you encode messages in a pre-defined way, like deep-space satellites do, you can cram in a lot more data. 

Things will however likely remain minimally human-readable. If you have to constantly look up things in a table, that just inhibits adoption. Plus, if you use ASCII characters as values, human-readable abbreviations are not that much bigger in message size. 

Binary or ASCII base compression messages is also a possibility. We'll see how things pan out.

viewsat, 2018-01-04:
@chamnit  May I ask you about the ARM version of Grbl its release time?

chamnit, 2018-01-04:
Soon-ish for a beta release. I had a NASA project since last June that just completed. It took all of my time. I have been getting reacquainted with the code, but have a lot to do.

closed 2018-02-02 by chamnit