gnea/grbl #344: coreXY movement direction in GRBL1.1

domenic-d, 2018-01-05:
Just updating from GRBL0.9 to 1.1 and we are having a problem achieving the desired axis direction movements.

Origin is in the bottom left (x=0, y=0)
Desired movement is
Left to right = positive x
Botton to top = positive y

We have tested all $3 combinations as well as changes to motor wiring without being able to achieve the desired directions of movement. When x is correct, y is inverted, and viseversa.

In GRBL0.9, this worked correctly with $3=2.

BTW, we are able to achieve the correct homing direction by setting $23=2. This works as expected.

Any suggestions on what we can try?

chamnit, 2018-01-05:
Hi Domenic. From what I recall, you were using your own version of Grbl v0.9. There may be difference between your version and the mainstream v0.9 and v1.1. I don't think I changed anything with v0.9 and v1.1. I also remember that your CoreXY setup was not what is described in the theory page. Your motors are in a different location. I can't remember if we had to alter the CoreXY code for you or not.

I just know that the CoreXY setup must be exact, starting with base motion. The motors must rotate in the direction shown in the theory with the direction of the axis. For example, for a +x motion, both A and B motors must rotate counter-clockwise. For a +y motion, A rotates counter-clockwise and B is clockwise. If they don't, then you'll have problems. You'll likely have to alter the CoreXY calculation code to compensate for the difference. It's usually just a matter of flipping sign of one or two of the calculations.

domenic-d, 2018-01-08:
Hi Chamnit,
Thanks for the reply.
Our setup is exactly as the default corexy configuration. Motors and belts are in the same place and movement is exactly as per the equations of motion.

The only difference is that our setup requires both deltaX and deltaY to be reversed from the diagram attached, because our machine is rotated 180degrees.

I will experiment with the equations and see whether we can achieve this result.

domenic-d, 2018-01-09:
Resolved. Thank you.

closed 2018-01-09 by domenic-d