gnea/grbl #348: Limit switch

NODOOD, 2018-01-10:
It would be useful if it reports the axis on which the limit switch was hit.

chamnit, 2018-01-10:
Status reports show limit switch state. If you mean hard limits, the AVR pin change interrupt does not contain information on which limit triggered the interrupt. Polling the switches inside the interrupt is not reliable.

closed 2018-01-10 by chamnit

NODOOD, 2018-01-11:
Ok, thanks

NODOOD, 2018-01-11:
Now that I have thought of it, it is possible to do so.
Because during the homing process, the MCU is capable of detecting on which axis the limit has been hit.
So of course, I have written a code in the firmware that tells me what axis has been hit.
Is there a way that I can add it to the real source code?
As a contribution to grbl?

chamnit, 2018-01-11:
There is nothing that tells you which switch triggered a pin change interrupt. That is intrinsic to the AVR design. Again, polling the state of a switch is NOT reliable, especially if a switch is actively bouncing, which it often is.

NODOOD, 2018-01-12:
OK, I get it. But it can be done using the same method that is used during the homing process. 
All you have to do is set a flag so that when the ISR is exited, the report can be done. Plus debouncing can also be done once the ISR has been exited.
So it is possible.

chamnit, 2018-01-12:
Yes, but you don't need to do debouncing after the ISR has exited. Grbl already reports the switch state in the status report. Also, Grbl already has a software debouncing option (using the watchdog timer as a spare timer), but it's disabled due to memory and flash constraints.

chamnit, 2018-01-12:
Also, homing is a different process altogether and operates in a special mode to allow it to detect the switch reliably by polling very very fast. Normal operation has to deal with comm, parsing, and planning in real time.

NODOOD, 2018-01-12:
Ok, understood.
Thank you