gnea/grbl #352: Question: Best practice for machine origin after homing

domenic-d, 2018-01-15:
Our machine uses CoreXY. It homes in the top-right of the workspace.
$3=0 & $23=0 yield correct direction of travel for movement and homing.

Upon 'homing' the top-right becomes 0,0 to GRBL. We have been defining a G54 to ensure the 0,0 is located in the bottom-left: G10 L2 P1 X-300 Y-210

During a job, if ctrl-x is issued to immediately 'stop' OR if the movement is 'paused' then stopped, it appears that position is lost and a re-homing is required.

This is contrary to the documentation "If reset while in not motion, position is retained and re-homing is not required."

Is there something we are missing or incorrectly doing with our approach above?
Is there a better way than using WCO to define the 0,0 origin that we would like?

chamnit, 2018-01-15:
Feed holding then issuing a soft reset should not invoke an alarm. It requires the hold to complete and come to a stop before the reset.

domenic-d, 2018-01-15:
This is correct.
No alarm is issued.

The problem we are having is that it appears position is lost after a soft reset as 0,0 is no longer correct.

chamnit, 2018-01-15:
This is probably due to your corexy axes being nonstandard. It should work fine on standard builds.

domenic-d, 2018-01-15:
The only change we made was to change the signs on the two coreXY equations (+ to -, and - to plus).
Our axes are standard but the motors are at the rear of the machine, which means is is simply rotated 180degrees from the images show for coreXY defaults. This sign chance allowed us to move in the correct direction.

Do you have any other suggestion on what we can look into for this issue?

biasedlogic, 2018-01-15:
Afair if I issue a soft reset, the machine goes into lock. I can either home it, or unlock. The last position is actually preserved, but it is my responsibility to check/confirm that the machine has not been moved in between

chamnit, 2018-01-15:
@domenic-d : There is more than one place the Core XY calculation takes place. Particularly when converting from position to steps.

domenic-d, 2018-01-15:
@chamnit @biasedlogic Thank you for sharing your insights on this problem we are experiencing.
I don't think we will find the answer here so I am closing this issue.

closed 2018-01-15 by domenic-d

chamnit, 2018-01-15:
I just told you the answer. Search for other places where corexy code is enabled. Particularly where position is reverted to steps.