gnea/grbl #353: Motion pausing on nodes

sinfocomp, 2018-01-15:
im running a 180,000 line gcode file at 7mm/s and after aprox 40 minutes motion stops being smotth and starts pausing at each node for a few ms, causing dots on the marking, im using denvi candle as sender app, if i abort, home axis and resend from current line starts smoth moving again and after another 40 minutes or so it starts the issue again, please help.

chamnit, 2018-01-15:
Please post your Grbl $$ settings and $I build info output. Stuttering can occur if your settings are too aggressive or if your tool paths are abnormally too high precision (i.e. 0.01mm per motion).

langwadt, 2018-01-15:
I'd suspect a memory leak or similar in the the sender, check the memory and cpu usage when it starts to go wrong

sinfocomp, 2018-01-15:
Settings
$$ < $0=10
$1=255
$2=0
$3=111
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.006
$13=0
$20=0
$21=0
$22=1
$23=11
$24=100.000
$25=3000.000
$26=232
$27=3.000
$30=100
$31=0
$32=1
$100=160.000
$101=160.000
$102=160.000
$110=6000.000
$111=6000.000
$112=6000.000
$120=3000.000
$121=3000.000
$122=3000.000
$130=1200.000
$131=2440.000
$132=50.000

BUild info
$I < [VER:1.1f.20170131:]
[OPT:V,15,128]

sinfocomp, 2018-01-15:
attached gcode file
[Fajo aros 44 garcia.txt](https://github.com/gnea/grbl/files/1633080/Fajo.aros.44.garcia.txt)

chamnit, 2018-02-02:
@sinfocomp : Sorry about the late reply. Forgot about this issues thread. I don't see anything unusual about your settings or g-code file. I think @langwadt may be right. It's something to do with your GUI. It's a pretty long file, so if it's not handled correctly by the GUI, it could lead to performance issues.

closed 2018-02-02 by chamnit