gnea/grbl #355: checksum algorithm returns boolean instead of byte

ChristianRinn, 2018-01-18:
Typical use of a boolean operator ('||') where the bitwise operator was meant ('|'). Fortunately there are compiler warnings.
```
grbl/eeprom.c:133:26: warning: '<<' in boolean context, did you mean '<' ? [-Wint-in-bool-context]
     checksum = (checksum << 1) || (checksum >> 7);
                ~~~~~~~~~~^~~~~
grbl/eeprom.c: In function 'memcpy_from_eeprom_with_checksum':
grbl/eeprom.c:144:26: warning: '<<' in boolean context, did you mean '<' ? [-Wint-in-bool-context]
     checksum = (checksum << 1) || (checksum >> 7);
                ~~~~~~~~~~^~~~~
```
Affected are the functions memcpy_to_eeprom_with_checksum() and memcpy_from_eeprom_with_checksum() here:  https://github.com/gnea/grbl/blob/master/grbl/eeprom.c#L130

chamnit, 2018-01-18:
Yep. I found this a couple years ago and came to the same conclusion that it would wipe every EEPROM since the checksum would be wrong. Decided to leave it alone until the next major version.

closed 2018-01-18 by chamnit

micromouseonline, 2019-05-20:
Perhaps a comment in the source code pointing to this issue would help others (like me) from 're-discovering' this error periodically?