gnea/grbl #358: coordinates x y z always on 0.000

karborg, 2018-01-22:
Hi helpers,

I'm nothing more than a newbie. I read as much as possible on this site to find how to solve the issue. Hopeless.

The issue:  Machine position and work position always stay on zero, even when I send a file to the arduino interface. At this time, I only connected 3 stepper motors. Nothing else. Stepper motors run ok whith manual commands, gcode manual commands and when a gcode file is sent to arduino. The only problem appening is positions staying on zero. Command $# returns zeros on all positions.

I tried with grbl command and with candle softwares. No way.

Does it need some initialisation string to run ?

I don't know if I give you sufficient informations to let you help me.

Thank you

karborg, 2018-01-22:
Could someone send me a piece of gcode that for sure, let coordinates change ?

karborg, 2018-01-22:
Thank you

You pointed the problem. I typed twice the ? and I was aswered:
? < ok
? < ok
as you see, the query gives a blank response.  I don't understand why, and I don't know how to restore a normal behaviour.

Maybe some special reset ? (memory - eeprom ...)

chamnit, 2018-01-22:
As @neilferreri suggested, check if your GUI supports Grbl v1.1. Or try a different GUI. You can connect to Grbl directly through a terminal, and you'll find that everything works fine.

neilferreri, 2018-01-22:
It's your GUI, candle. You are using one that is incompatible with your version of grbl.

karborg, 2018-01-22:
To neilferreri:

Candle version 1.1.7. All other commands are ok. The same issue occurs with grbl controller V3.6.1

karborg, 2018-01-22:
Ok I'll check.

Thank you

jahnj0584, 2018-01-22:
G90 G21
G0 X5 Y5 Z5
M30

karborg, 2018-01-22:
Well, I'm peeping out the fog.
I sent g1 x5 y5 z5 from arduino serial terminal, then a ? and it answered <Idle|WPos:5.000,5.000,5.000|Bf:15,127|FS:0,0>
You were right: the problem resides on GUI software. Coordinates are computed correctly.

If you agree with this, we can close the post.

Thanks a lot

karborg, 2018-01-22:
Just before leaving: can someone advice me another GUI compatible with grbl 1.1 ?

jahnj0584, 2018-01-22:
Universal Gcode sender 2.0 platform or Sourcerabbit

neilferreri, 2018-01-22:
I like CNCjs. 1.9.11

X3msnake, 2018-01-22:
BCNC

and check the wiki

https://github.com/gnea/grbl/wiki/Using-Grbl

karborg, 2018-01-22:
Thank you, both work !

zeevy, 2018-01-23:
If you can attach Bluetooth to your machine you can try mobile version.
https://play.google.com/store/apps/details?id=in.co.gorest.grblcontroller