gnea/grbl #359: bcnc: Alarm6

psyqjo, 2018-01-23:
After the conversion of my router I suddenly have the problem that after the homing of the Z axis always the alarm 6 is displayed. why can this be? I have already used another limit switch but exactly the same problem.

chamnit, 2018-01-23:
According to the [wiki page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#alarm-message), this alarm message is thrown when the homing cycle is aborted with a soft-reset. This either means you are some how sending ctrl-x commands or the soft-reset pin on the Arduino is getting enough EMI to trigger. The latter is the likely culprit.

closed 2018-01-23 by chamnit

psyqjo, 2018-01-23:
i had the alarm on an other arduino too. but witch is the soft reset pin?
and can it be that the error is set by an short cut on pcp?

chamnit, 2018-01-23:
These are the only two events that can trigger an Alarm:6. I suggest looking into them. 

There is a pin diagram in the wiki pages.

psyqjo, 2018-01-23:
okay have the error, had a short circuit at the pin D12 and A0, there was a barely visible solder bridge. everything is possible now.