gnea/grbl #360: Latest grbl_v1.1f.20170801  not working

mubarak111nsu, 2018-01-24:
Dear respected sir, Previously, I used grbl_v0_9a_edge_328p_16mhz_9600_build20121210 . It works perfectly.
But after uploading latest version, it is not working at all.
It shows
**** Connected to COM4 @ 115200 baud ****

Grbl 1.1f ['$' for help]

and when, I execute, $$ then
it shows
>>> $$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
not like all other description that I seen previous version.

If I click on Machine Control, -X  or +X or -Y +Y
then it does not move or work at all.
Please could anyone help me?
I will be grateful to you.

zeevy, 2018-01-24:
the GUI you are using may be not compatible with grbl 1.1f
lot of things changed in 1.1

mubarak111nsu, 2018-01-24:
Dear respected sir, I'm really happy for your reply.
Please could you help me to solve the problem?

X3msnake, 2018-01-24:
Chose a different sender from the available ones listed on the wiki

https://github.com/gnea/grbl/wiki/Using-Grbl

Download the lates version of that sender for example bcnc or ugs

Try connecting and see if it works

No dia quarta-feira, 24 de janeiro de 2018, Muhammad Mubarak Hossainمحمد

closed 2018-02-02 by chamnit

mubarak111nsu, 2018-02-03:
I have solved the problem.