gnea/grbl #362: Soft start spindle

iMaks-RS, 2018-01-25:
Hello!
Please, modify firmwire to soft start spindle.  This option need to resolve problem with power supply. 
If spindle rpm is hi, start current so mach and voltage down  from 24 to 4 Volts. If insert in G-Code rows
S300
S1000
S4000

Spindle is soft start and not problem with power supply.
I noob on C++, but know code howto write on writing-code.

May be create new option as dsp controller RZNC D5416 and VFD - time to start spindle from low to hi rpm. On DSP 5416 this options name "SPINDLE DELAY" in ms to max rpm.

Sorry for my english.
Best regards,
Maxim

chamnit, 2018-01-25:
In short, not going to happen. Reason, no room on the 328 AVR. All space and memory is literally maxed out for certain critical configurations. 

If you did a search of the closed issues, you would have found a solution on this issue. You can simulate a soft start by replacing your M3 gcode with a ramp up and dwells.

closed 2018-01-25 by chamnit

jahnj0584, 2018-01-25:
The VFD should also be in control of this.

iMaks-RS, 2018-01-25:
i have mega2560 to.. Realy create mod for this pcb?

mayhem2408, 2018-01-26:
@iMaks-RS . Even if there was a soft start to help you with your startup issue, you might find that the load while cutting might cause the same problem.
I'll have to find the details on what a friend did, but I know that they wrote a custom macro they simply replace M3 with something like M10003 and the custom macro handled the soft start. That way the GRBL controller does not have to do anything special.

iMaks-RS, 2018-01-26:
Sorry, but this is a crutch and not a solution to the problem. Yes, you can make it so that FlatCam automatically adds lines .. but if I need to change the environment? For example ArtCam or SolidCam? I always have to remember that I need to add a piece of code to my hands? Do not you find this "strange"?

I understand the problem with the size of the memory of Atemega328 ... But for Mega2560 it is quite possible to implement it, but enough memory.

Sorry .. but I do not understand the refusal to implement a soft start in Mega2560.

I'll try to do it myself.
In any case, thanks for the open source project and the source code.

Sincerely.

chamnit, 2018-01-26:
@iMaks-RS : I think you are missing the point. This is not a refusal. Sure the Mega2560 has the room, but I have larger priorities than to add a soft start feature for a small set of users that may need it. 

In practice, it really shouldn't be the job of Grbl to regulate the soft start. It's the job of the spindle implementation. There are infinite ways someone could have their spindle setup and infinite set of requirements unique to each of them. It's impossible for Grbl or any general purpose CNC controller to support everything. At best, you can support a majority, which could still happen in the future, but not anytime soon.

To simplify things, the only job Grbl and other CNC controllers should really do is tell the spindle when to turn on, how fast, and possibly a feed back mechanism for an error. This is why others are telling you that you can tweak how your system runs by either gcode via spindle speed and dwell commands (M3 Sxx; G4 Pxx by a simple text editor with a find and replace tool) or by implementing a separate controller custom for your setup to regulate the output from Grbl to your particular spindle setup. 

We do not know the conditions of your spindle setup and requirements. Any solution that we come up with will not likely be optimal for your situation. It's very simple to take another Arduino and program it to interpret Grbl's output and convert it for your particular situation. That's much easier than directly modifying and hoping that you didn't break something important.

chamnit, 2018-01-26:
In addition, I understand that you'd like this problem solve by software, and we have given you several options for this. But, you really should be looking into directly solving the source of your issues, your hardware and power supply. There are lots of ways to handle power surges, i.e. a large capacitor to buffer the large initial energy draw. You also need to find out whether or not a soft start will even help your situation too. Either way, you've got a lot of work you need to do.

iMaks-RS, 2018-01-26:
"a small set of users" is all owners of a Chinese miracle based on a woodpeaker board. This board has a very idiot design of the power system. The interface and mega328 are powered from the usb port, in case of a whine or vibration - the connection is lost .. this has already been verified once.

I'm not asking you to arm yourself with an interpreter and read the thread of the forum on 2418 on the Russian resource .. I just ask you to add a simple loop that will stupidly increase the value of pwm from 0 to 254 in 3-5 seconds. As I understand from the spindle.h file, the parser does not process the code until there is a status about starting the spindle. What you can do for 10 minutes .. I'll do it in 2-3 days .. but I'll do it .. I'll have to read and watch training videos ..

do not necessarily add this to the project, just share the code or the modified file, at my own risk. I just started thinking about designing my own controller board with the correct power system and opto-isolation.

chamnit, 2018-01-26:
@iMaks-RS : Please calm yourself down. Yes, a "small set of users". Woodpecker boards have recently become more popular now but they are not the only Grbl board in existence, nor are they the majority, nor do most people wire in a spindle. You also have to remember that these boards are cheap for a reason. In my experience, you get what you pay for and have to buy things to fix them.

Seriously though, is it really that hard to use a text editor? All you need to do is find every M3 command and replace it with a series of:
``` 
M3 S1
G4 P0.5
S50
G4 P0.5
S100
...
```
This will essentially do the same thing without you having to learn code.

iMaks-RS, 2018-01-26:
Ah ... yes .. how .. "capitalism" .. sorry. Have a nice day.