gnea/grbl #364: 4th Axis for foam cutter - please, please pretty please :)

mumin50, 2018-01-26:
Hi all,

I started to build 4 axis cnc foam hot wire cutter, and got cnc shield v3. Looked good 4 drivers for stepper cheap, but... 
I can't get GRBL to work with 4th axis
Help :)

In total i would need 
4axis xyza
4stops xyxa
1 wire enable best on PWM for mosfet
1 imput for temp read, or pot to regulate PWM for wire

Any advise

X3msnake, 2018-01-26:
GRBL in the UNO will only supports 3 Axis, no more memory space for any
features.

You need to look into GrblMega 4 Axis  ot TinyG

https://github.com/gnea/grbl-Mega/issues/15
https://github.com/gnea/grbl-Mega/pull/21
https://github.com/synthetos/TinyG

mumin50, 2018-01-26:
ok than i need to take arduino mega and just connect it to cnc shild v3, or get ramps.
Pin-out i will find in cpu_map (info for all that search after me)

One thing strange is that probably all would fit inside UNO if other things would be not there: spindle, etc :) but whatever 5$ will be ok and finding things is more fun.

X3msnake, 2018-01-26:
Maybe there is already a uno project for hot wire cutting
Good thing about OpenSource is that if you find that there isn't you can
always share back a modified version of GRBL

Found this btw
https://sites.google.com/site/foamcuttercnc/software
https://sites.google.com/site/foamcuttercnc/electronics

closed 2018-01-26 by chamnit