gnea/grbl #365: Trinamic TMC2130 Stepper Driver support

Blaisun, 2018-01-26:
Has there been any discussion on support for the TMC2130s? 

 In my brief research, they seem to have a number of benefits over the 4988 and 8825 drivers. The ones i am most interested in are dynamic current control and stall detection.  i am not sure how well the "virtualized endstops" will function in a CNC application but they seem to work well for homing in a 3D Printing application.

The biggest downside i can see for use in grbl is the pin shortage on the Uno. Would it be feasible to implement on the mega branch?

This is more of a question of feasibility directly related to grbl as it currently seems the project doesnt support anything other than Step/Dir for driver control....

lslewis901, 2018-01-26:
Here is a link where a guy used another Nano to integrate the features with the TMC2130 drivers:

https://youtu.be/ZzO9skEYqwY
Wiring and code page: https://reps.cc/?p=5

chamnit, 2018-01-26:
Support may show up on the Mega branch at some point, but it's not a priority right now.

closed 2018-01-26 by chamnit

Schildkroet, 2018-01-26:
watterott offers some replacement modules for A4988 etc. with TMC chips.
https://github.com/watterott/SilentStepStick

Blaisun, 2018-01-27:
@chamnit  ok thanks. i was just wondering if the requirement to talk to the chips via SPI would mess with the way grbl currently works....

Rotzbua, 2018-03-31:
The TMC2130 supports 3 modes how the chip can be run.
One mode is the "standalone mode" which can be configured via pins and require no SPI. In standalone mode the TMC2130 can replace a A4988 without any software modification.

Result:
It is possible to use 2 features of the TMC2130 with this library: stealthChop and spreadCycle. Basically this are the features of the TMC2100 because we do not use the SPI.

dudman, 2018-11-25:
Sorry for the late reply, been searching everywere else. 

So the 3rd feature of TMC2130 SPI is not implemented in Grbl ? Is there any plan to support this ?