gnea/grbl #366: Esp32 port ?

mumin50, 2018-01-26:
Hi 
I was thinking if it would be not possible to port it to ESP32, including WiFi communication, enough pins for even 5 axis, cheap, arduino programmable, hips of memory, fast...
I'm not C programmer, but would gladly look in arduino code

X3msnake, 2018-01-26:
Check the Arm Port Discussion

https://github.com/gnea/grbl/issues/67#issuecomment-360717360

chamnit, 2018-01-26:
There are a lot of ARM processors that Grbl can run on. ESP32 is one of them. I will not support it directly, but the new upcoming abstraction layer should facilitate someone porting and supporting ESP32 easily.

closed 2018-01-26 by chamnit

luc-github, 2018-01-26:
@chamnit may I know where I can see the code of this new abstraction layer?  thank you

chamnit, 2018-01-26:
@luc-github : It has not been released yet and won't be until it's ready. This will help prevent multiple versions of the abstraction layer being out in the wild. Things are still pretty fluid and I'd like to avoid painting myself into a corner, so to speak.

luc-github, 2018-01-26:
@chamnit understood thank you for doing this, you doing great. keep going ^_^

anlpereira, 2018-06-23:
Hi @luc-github, is it ready the grbl to ESP32 board? Thanks a million.

luc-github, 2018-06-23:
I am not working on it on my side - and I do not know if anyone did it already 
the last status I get was the one above : waiting for the new abstraction layer of grbl

cri-s, 2018-06-23:
As I recall ramps hal works on esp32 arduino port. But its only 0.9 and not
1.1 as I know.

rgwood-github, 2018-07-26:
https://www.youtube.com/watch?v=A9TNu58Sq00

luc-github, 2018-07-26:
Nice ^_^

Harvie, 2019-03-16:
> There are a lot of ARM processors that Grbl can run on. ESP32 is one of them.

I think that ESP32 is new architecture designed from scratch rather than ARM based. At least i was not able to find any mention about ESP32 being ARM.