gnea/grbl #367: Probing problems

JeroenVanDenHorn, 2018-01-28:
I'm running into issues with auto-leveling on 1.1f. The G38.2 command works fine initially. It moves the probe and lowers it until it makes contact. Once contact is made GRBL no longer responds to commands. When using the auto-leveler in chilipeppr's jgradie workspace the leveling stops. Hitting pause twice makes the commands appear in the console but the CNC does not react. The only thing I can do is to detach the probe, hit pause twice and quickly reattach the probe. Which is quite cumbersome when leveling 180+ points... 

I have a feeling that all stepper commands are being blocked when the probe is active. I will in the mean time create some electronics that will send a short pulse on contact (10 ms) instead of maintaining the active 'LOW' state. But it would be nice to have this fixed.

langwadt, 2018-01-28:
I find it hard to believe that there would be a problem with probing in GRBL that needs fixing, lots people use it all the time. Suspect hardware problems with noise

JeroenVanDenHorn, 2018-01-28:
You just beat me to updating my own issue. The probe pin was not connected correctly causing the entire arduino to stop. That meant that the probe stopped on contact but not for the correct reason. I traced the A5 pin back to the shield and everything now appears to be working correctly. I jumped to this conclusion because with 0.9 there also were issues where you had to pause/resume on every probe.

closed 2018-01-28 by JeroenVanDenHorn

timmit99, 2018-02-23:
What was your solution? I am having the same issue on my CNC Shield V3.000 and would love to solveit