gnea/grbl #368: Pin 12 not responding

oliveat, 2018-01-28:
the Z limit input [12] is not responding. The wiring is fine. The fault occurs with either of the two arduinos I use. I tried to find the config.h file that would allow me to change the homing cycle to exclude z axis homing but have, under the search, a multitude of such files and cannot work out which has the C language to allow the modification, Richard

jahnj0584, 2018-01-28:
It's in config.h. It explains inside the document how to do 2 axis homing.
Sure your switch works?

oliveat, 2018-01-28:
Thanks for the reply, I am certain the switch works, I have traced it all the way to pin 12 on the uno. please be more explicit as to how I find the relevant config.h file, Richard

X3msnake, 2018-01-29:
https://github.com/grbl/grbl/blob/master/grbl/config.h

Line 60 to 77

oliveat, 2018-01-30:
Thank you X3, just the ticket

X3msnake, 2018-01-30:
We assume you are compiling from the Arduino IDE. The file you must change
is in the library

oliveat, 2018-01-30:
Yes ,the Arduino IDE. My initial confusion was how to edit the file. Once I worked out I needed a script editor to modify the config.h file I turned off Z axis homing and was able to get the system running. The reference you suggested will allow me to use input pin 11 as the Z homing/limit input if I re-enablle Z homing as a separate event in the config.h file. I think that is an elegant solution so Thank you.

109JB, 2018-01-30:
any text editor can be used to edit the config.h file

oliveat, 2018-01-31:
Hi again, what do I change in cpu_map.h to use the x limit/homing pin for z limit/homing also in lines 58 to 68?

oliveat, 2018-02-01:
AT LAST THE ANSWER!! After a steep learning curve in the syntax of Arduino and careful analysis of the code in config.h, cpu_map.h, and the cnc shield I own I discovered that pin 12 is actually connected to spindle enable and pin 11 is connected to z axis limit. When I reset the defaults in the .h files and connected theZ limit switch  to spin enable, the limit and homing worked for X, Y AND Z Thanks every one for your help, Richard

closed 2018-02-01 by oliveat

X3msnake, 2018-02-01:
Good to know you found a solution, thanks for sharing back :)

Happy machining ;)