gnea/grbl #370: Soft Limits, Home switches, nothing works

printmyplastic, 2018-02-01:
I've tried everything. RC filters, opto-isolators, mechanical relays, both NO and NC switches, both external pull up and pull down resistors, multiple iterations of grbl including the grbl mega port; I simply cannot get anything homing related to work with grbl. Right now I figured that mechanical relay + NO switch would be the easiest to get working, because the audible click would at least let me know that the switch is functioning even when grbl fails to read the input. I have worked with a lot of hobbyist diy electronics stuff, but I'm afraid there is just something I'm doing completely fundamentally wrong. Same exact wiring works flawlessly (for homing) with a beaglebone, but I simply cannot get homes switches to work with grbl no matter what I do. All wiring is verified to work correctly with a multimeter. I'm using a bare arduino with no shield. The following are my settings, but I've tried with Hard limits both on and off, soft limits on and off, and it makes no difference. I'm using a bare arduino with no shield. Willing to try anything.

$0=10
$1=25
$2=0
$3=4
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=1
$20=1
$21=1
$22=1
$23=4
$24=25.000
$25=100.000
$26=25
$27=3.000
$30=1000
$31=0
$32=0
$100=83.485
$101=83.485
$102=83.485
$110=3000.000
$111=3000.000
$112=2000.000
$120=10.000
$121=10.000
$122=10.000
$130=1200.000
$131=2400.000
$132=600.000

chamnit, 2018-02-01:
What kind of "error:x" messages are you getting? They provide clues to what's going on.

printmyplastic, 2018-02-01:
Right now, with this particular setup (relays + NO switches) I get no error. It just ignores the stops completely. Both the 328 and 2560 grbl do this. But I just tried running the router off marlin, exact same setup, homing works perfectly. I must be half retarded to not be able to figure out what this problem is. I'd much prefer to be able to use real standard Gcode, but I really don't even know where to start with this issue, I've spent hours if not days trying to diagnose it. I have an oscilloscope so I know the issue is not noise.

chamnit, 2018-02-01:
You can check the status reports by connecting through a terminal and issuing a ? character. Grbl v1.1 will show if a switch is detected as triggered in a report. I would check that first. Also note that v1.1 and v0.9 have the z limit pin on D11 not D12, if you are referring to old docs.

chamnit, 2018-02-01:
Oops I meant z limit is on pin D12 not D11. Not the other way around.

closed 2018-02-14 by chamnit