gnea/grbl #371: Real time status report current feed rate report zero when moving

arkypita, 2018-02-01:
Hi all, I am going to add current F and S value to my LaserGRBL UI.
F and S are decoded from real time status message as documented here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports

I made some test with manual sent movement (like G1 X100 Y100 F200) and it shows the current F and S value, so parsing data is ok in my code.

However if the gcode is composed by a long list of very small movement (like when I stream a raster image encoded in a lot of segments) the feed rate F reported by grbl is always zero.

[image]

I don't know if it is an issue or it is "by design" but i report it.

radiolee, 2018-02-02:
@arkypita mcu too busy and drop the request? is it any report back after send "?" ?

arkypita, 2018-02-02:
@radiolee 

The report are sent continuously, the S value is reported correct, but the F value is always zero.
You can see the output in Debug windows (Lower-right of the screenshot. The screenshoot was taken during movement).

chamnit, 2018-02-02:
@arkypita : This may happen if the streaming is starving the planner buffer. Is the streaming rate exceeding about 250 lines/sec? Or is the motion stuttering?

arkypita, 2018-02-02:
- streaming rate is not so fast
- streaming using send-response or using 128 char count protocol doesn't matter
- ? query frequency doesn't matter

It happens only if LaserMode is OFF and with frequent S changes. I think it is because with LaserMode OFF grbl must stop to change spindle speed, it is correct? With constant S value or with LaserMode ON the behaviour is ok.

Video: https://www.youtube.com/watch?v=K-WYIZiylYk
debug show all the status message that carry FS: information

chamnit, 2018-02-02:
Ok. So it is stuttering, except intentionally with laser mode off, which requires a stop in motion when changing S values. I put a note for future development to try averaging the current rate over a segment, rather than instantaneous rate computed at the end of a motion as it does now. This is why its always zero. Unfortunately, I'm leery of asking the AVR to do unnecessary floating point calculations like averaging over time, so it'll have to wait for the ARM version to address this.

closed 2018-02-02 by chamnit