gnea/grbl #372: homing does not work

cecconilorenzo, 2018-02-02:
Hi, I'm trying to put the limit switches for homing but nothing works.
the problem is not the wiring but the software, I have enabled $ 22 = 1 and when I send the command $ H the motors x y are blocked.
I do not have a Z axis because it's a laser cnc, is this a problem?

chamnit, 2018-02-02:
@cecconilorenzo : Yes. Grbl is setup of 3-axis. Not 2-axis. If you did a search for this, you would have found lots of issues threads detailing how you would alter Grbl to home for 2-axes. In short, you need to edit your Grbl config.h in the Arduino library. There are instruction in there.

closed 2018-02-02 by chamnit

cecconilorenzo, 2018-02-02:
OK thank you very much

montakhab, 2019-07-23:
HI. I began to UGS  platform  v2 nightly setup wizard but in home settings when i press the homing button
occurs error. Whats the problem?
[image]