gnea/grbl #373: Pause waiting for limit sensor during job

AAKOTb, 2018-02-04:
Hello. I would like to add functionality to the library GRBL I design cnc embroidery machine now. To work with the embroidery machine, it is necessary to stop the program until the needle of the sewing machine rises. Is it possible to implement the function of stopping the code in the library, until the information is obtained from sensor the needle of the sewing machine? As a needle position sensor of the sewing machine, I would like to use one of the limit sensors.
So if our code looks like this:
N10G01X0.000Y0.000
N11G01X2.622Y0.744
N12G01X2.622Y0.744
N13G01X2.657Y0.709
N14G01X2.622Y0.673
N15G01X2.657Y0.638
N16G01X2.622Y0.602
N17G01X2.657Y0.567
N18G01X2.622Y0.531
N19G01X2.693Y0.602
N20G01X2.728Y0.567
N21G01X2.764Y0.602

That machine will perform the first line N10G01X0.000Y0.000 and wait for information from the needle position sensor. After triggering the sensor, proceed to the second line N11G01X2.622Y0.744 and so on.

cheton, 2018-02-04:
This should be controlled from the G-code sender. You can modify [simple_stream.py](https://github.com/gnea/grbl/blob/master/doc/script/simple_stream.py) to fit your needs.

AAKOTb, 2018-02-04:
And what about the functions of the M66? I found information about this function here
https://www.shapeoko.com/wiki/index.php/G-Code
M66 (M66: (EMC2) Input control)
This is exactly what I need. I can put the function M66 P11 L1 Q100000 after each line in the g-code and connect to my pin 11 my needle position sensor. But is this realized in GRBL?

zeevy, 2018-02-04:
You can try placing hold "!" After each line and connect your sensor to cycle start pin.

patgadget, 2018-02-12:
Can a delay work for you?
G4

patgadget, 2018-02-12:
Or you could use the Probing G38.2 method also

closed 2018-02-14 by chamnit