gnea/grbl #374: HAL for grbl

pmbdk, 2018-02-05:
In the FAQ there is a mention of a HAL for grbl and I have also seen it mentioned in one of the discussions here. Since I am in the process of making a port for a Raspberry Pi/FPGA combination, I am very interested in hearing about this (or just if there is a template I could follow so it will be easier to port once the HAL is complete?).

chamnit, 2018-02-05:
Hang tight. The HAL is still in a fluid state and I don't want to release anything now to prevent multiple versions of it being out there. Don't worry I was able to do a lot of work on it this weekend. It's coming along nicely.

closed 2018-02-05 by chamnit

holla2040, 2018-06-21:
Any HAL progress? just curious, I'm thinking about running on a Cypress PSoC5. I've been pouring through a previous port which touches just about everything. HAL would be awesome! Thanks.

chamnit, 2018-06-25:
The HAL is pretty much complete, but have been working on new features (new planner). It'll still be a while I think. I seem to keep coming up with more things I should do. I probably need to stop and save some for later.

ghartc, 2018-06-25:
Awesome - can you describe your new planner!?

Maybe you want to pre-release it before you have too many new features to
debug! ;-)

Gregg <https://www.facebook.com/pages/Carlson-Design/123978984306367>

holla2040, 2018-06-25:
Thanks for the update.  I'll check back in 6 months or so and see how you're doing. Thanks for the effort, we all love grbl.

Schildkroet, 2018-06-25:
Good news, i hope it won't take that long until release

chamnit, 2018-06-25:
The planner has true 3rd order motion control. The mathematics are complete, which were far from trivial, and going through virtual testing and optimization now. I'm also looking into special planner modes that will allow different functionality and customizations for specific end use cases. This I won't elaborate on.

ghartc, 2018-06-25:
No, it's not trivial, but going to be interesting! Was TinyG useful, or did
you approach it differently (or find a flaw there)? Will you have time to
publish the math?

Thanks!

Gregg Carlson
CarlsonDesign.com
TristarGlass.com
539 S Trenton Ave
Tulsa, OK 74120

www.carlsondesign.com <http://www.carlsondesign.com/logo+green+40.bmp>

<https://www.facebook.com/pages/Carlson-Design/123978984306367>

chamnit, 2018-06-25:
Unless TinyG has changed recently, it performs a series of approximations to compute its plan and then fits higher-order curves into it. As a result, the plan isn't guaranteed to be optimal, let alone stable. There are a lot of short cuts and edge case checks to get it to work reasonably well.

The main thing I've done is compute the exact closed-form and optimal solutions in a way that isn't as computationally expensive as they could be. It's still expensive, but I'm fairly confident that it'll work in realtime on an ARM processor. Also, as with Grbl, the key to making this all work is the junction deviation model. This solution was by far the hardest to get exactly right and remain cheap to compute. 

I may publish the math on all of this, but probably a bit of time after the code's release to keep the project unique for a while.

X3msnake, 2018-06-26:
Awsome :)
You are the Master Sonny thanks for all your hard work

Harvie, 2019-03-03:
Pardon me for dumb question, but what exactly is "HAL for GRBL"?

X3msnake, 2019-03-03:
Hardwares Abstraction Layer

means you can compile to any uc not just arduino

Harvie, 2019-03-03:
> means you can compile to any uc not just arduino

Very nice! Thanks for explanation. It would be cool if there also was a HAL for peripherals.

So it would be very easy to modify GRBL from using step/dir stepper drivers to servos. Or rs485 for spindle instead of PWM. Etc... There are many ports of GRBL that are made just for doing stuff like this. If there was abstraction layer for this stuff, it might be easier to maintain such forks.