gnea/grbl #375: Issue on leading zeros in read_float function

setaperlacloche, 2018-02-09:
All the following G-Code lines :
```
- G0X1
- G0X1.00000000
- G0X000000001
```
are identical. But in Grbl the last one is misunderstood and becomes `G0X0`.
According to G-Code specifications ([http://ws680.nist.gov/publication/get_pdf.cfm?pub_id=823374](http://ws680.nist.gov/publication/get_pdf.cfm?pub_id=823374) p.17) :

_"Notice that initial (before the decimal point and the first non-zero digit) and trailing (after the
decimal point and the last non-zero digit) zeros are allowed but not required. A number written
with initial or trailing zeros will have the same value when it is read as if the extra zeros were not
there."_

Adding leading zero tolerance can be done by inserting in `read_float` function in `nuts_bolts.c:51` the following code :
```
  // Skip leading zeros
  while(c == '0') {
    c = *ptr++;
  }
```

What do you think ?

chamnit, 2018-02-09:
@setaperlacloche : The third example has more than 8 digits. Grbl's strtof function assumes that you will have 8 or less.

closed 2018-02-09 by chamnit

setaperlacloche, 2018-02-10:
I see, it's by design.

So is it possible to :

- Add in wiki (perhaps in section Home/Limitations by design) some words to warn the user about this limitation.
- Return an error (Error:2) instead of running a badly interpreted command.

chamnit, 2018-02-10:
I suppose. But the NIST document doesn’t state how many leading zeros or total number of digits are allowed. In practice, no one will have 8 leading zeros before a non zero digit or decimal.

Also, Grbl’s string conversion function is a pared down version of the standard lib strtof() function. Designed to minimize overhead, size and maximize speed. All needed for a slow 8-bit processor. 

I won’t change this for Grbl on the AVR due to space and speed constraints but can update it for future versions.

biasedlogic, 2018-02-10:
Regardless whether it's gonna get changed or not, I'd see it in the code parser on the sanitisation level (where white space is stripped etc.). Wasteful to throw leading/trailing zeros at string to float conversion, and stripping leading zeros is easy.