gnea/grbl #376: Feature : Add ALARM code on status reports

setaperlacloche, 2018-02-10:
Currently only HOLD and DOOR states have a sub-state in Status Reports.
Is it possible to add the Alarm code when in Alarm state, in order to get a message like this :
```
<Alarm:3|MPos:10.172,14.476,3.000|FS:0,0>
      ^^
````

109JB, 2018-02-10:
It's already there

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#alarm-message

setaperlacloche, 2018-02-10:
In reply to `?` command, i.e. in Status Reports

chamnit, 2018-02-10:
Not a bad idea, but no changes to the interface will be made to the current version. This will break existing GUIs. 

I’ll add this to my development notes to address this in the next major version.

chamnit, 2018-02-10:
I also think there may be ways to uses the substrate idea to clarify other running states like Run. For example, a Run substrate could denote if the machine is executing a motion or dwell.

closed 2018-02-10 by chamnit