gnea/grbl #378: Cant upload after enabling CoreXY in latest build.

jcphlux, 2018-02-12:
If I change the config.h file to enable CoreXY I get an error on uploading. I am hoping it is just something simple I an doing wrong.

```
vrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7800
         0xff != 0x16
avrdude: verification error; content mismatch
```

I can revert the changes to the config.h and it uploads fine or if I just comment out line 189 `#define COREXY` it still uploads fine.

Here is the changes I made to the config.h

lines 104 to 113
```
// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable

// NOTE: The following are two examples to setup homing for 2-axis machines.
// #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle. 

#define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
#define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y
```

Lines 182 to 189
```
// Enable CoreXY kinematics. Use ONLY with CoreXY machines.
// IMPORTANT: If homing is enabled, you must reconfigure the homing cycle #defines above to
// #define HOMING_CYCLE_0 (1<<X_AXIS) and #define HOMING_CYCLE_1 (1<<Y_AXIS)
// NOTE: This configuration option alters the motion of the X and Y axes to principle of operation
// defined at (http://corexy.com/theory.html). Motors are assumed to positioned and wired exactly as
// described, if not, motions may move in strange directions. Grbl requires the CoreXY A and B motors
// have the same steps per mm internally.
#define COREXY // Default disabled. Uncomment to enable.
```

jcphlux, 2018-02-12:
I am getting the same issue on version 0.9. I have tried on 3 separate boards.

langwadt, 2018-02-12:
I'm guessing you have run out of memory, https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#flashing-grbl

jcphlux, 2018-02-13:
I took an extra Uno and made a ISP loader and was able to put a reload the boot loader and was able to load the build on it.

closed 2018-02-14 by chamnit