gnea/grbl #380: Endstop connecting

jarnoburger, 2018-02-16:
I am double checking : 
*I have a bunch of endstops. You can look at them here : 
https://archive.is/UwDFo
*I have a cnc shield. You can look at it here : 
https://blog.protoneer.co.nz/arduino-cnc-shield-v3-00-assembly-guide/#EndStops
*The fan and motors and power adapter are working.

My endstops have 3 pins : gnd , 5v , signal >>I think they are in N(ormally) O(pened) state.
The shield has 2 pins, so I deduct : 5v , receive signal (when closed).

I am going to test this , but I am not a electronics god :).
So.. If I would steal some gnd somewhere on the board,  it would work normally right ?

wlaw, 2018-02-16:
Be sure the board is using the default setting of signal to ground (low) for activation.
[image]

then using Pin 2 or 3 for X-, Y- and Z- 
and Pin 4 for X+, Y+ and Z+  should work fine.

in the switches normal state
Pin 1 = 1 (High) +5v
Pin 2 = Ground
Pin 3 = Ground
Pin 4 = 1 (High) +5v

in the switches switched state
Pin 1 = 1 (High) +5v
Pin 2 = Ground
Pin 3 = Ground
Pin 4 = 0 Low, Ground

Hope that helps

jarnoburger, 2018-02-16:
**Thanks to your answer , I connected**  
shield>5v           to switch>5v (pin1)
shield>X+ GND  to switch>GND (pin2/3)
shield>X+ SIG    to switch> SIG (pin4) 

**Then i update the status mask ($10) towards :** 
- Show Machine Coordinates (bit 0 > decimal    1) 
- Show Pins                             (bit 4 > decimal 16 )

Combining these two becomes: 1 + 16 = 17 
So I send code: $10=17
This changed my status mask to show machine coordinates and pins

**Time to check if the X thingie works :** 
I check the status code unpressed by sending character ?
My answer I get back is : <Idle|MPos:0.000,0.000,0.000|FS:0,0>

I check the status code while being pressed by sending character ?
My answer I get back is : <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:X>

So this part works !!
I can now connect 1 X and 1 Y towards that CNC v3 shield , but trying more will reset the Arduino.
I
I am now testing your dipswitch thingie , and why the other ports are still not working.
I will update my progress

jarnoburger, 2018-02-16:
I found that the pins beloning to SpnEn are the ones or Pn:Z

ednisley, 2018-02-18:
It turns out "Makerbot" style endstop PCBs connect the NC switch terminal directly to +5 V, in addition to having a pullup resistor on the C terminal. When you connect two endstops in parallel, tripping either switch crowbars the Arduino power supply: 

- From +5 V power supply
- NC terminal of the untripped switch
- Through the shared C terminals
- NO terminal of the tripped switch
- To power supply common

Which, of course, resets the Arduino and may give the Raspberry Pi's USB port something to worry about.

Snip the NC terminal from both switches and they'll work the way you (and I) expect ... [sigh]

More details, the schematic, and a post-surgery picture of an endstop on my blog at https://softsolder.com/2017/11/30/mpcnc-makerbot-style-endstop-switch-modification/

closed 2018-02-19 by chamnit