gnea/grbl #383: Computer not communicating with GRBL Arduino Nano Board

adigitaldj1, 2018-02-21:
Hello All!

I am a newbie to cnc but not Arduino. My problem is that I can't communicate with Adruino Nano GRBL board on my new cnc3018. I first of coarse tried running Universal Gcode Sender and it wouldn't communicate and would crash UGS. I had the correct comport set and set to 115200 baud rate. So I tried sending $ and $$ in Arduino Serial Monitor with no luck to pull up the help menu in there but I did get the $ for Help when the baud rate was set to 115200. I tried s new cable, nano board and another computer with no luck. I am running GRBL V1.1 and setting for comport at 115200.

Can someone please tell me what could be going on!

chamnit, 2018-02-21:
Verify that your Arduino Nano isn't broken for some reason. Try uploading an Arduino sketch on it to see if works. If so, then reupload Grbl on it.

adigitaldj1, 2018-02-21:
Yes tried a new Nano and tried my own code on it and was able to read data back on the serial monitor. I wasn't able to send through the serial monitor because my code I put on the Nano was not programmed for that.

Is there any test code for checking communications between a computer and Arduino?

What is the proper steps for flashing the Arduino Nano?

Is GRBL to run ok on a Nano?

chamnit, 2018-02-21:
You just did the test.

See the Wiki on how to flash Grbl. Follow them exactly. 

The standard Grbl build will fit on a Nano, but it'll be very close (within a handful of bytes) to full. Depending on the Arduino IDE version, the compiled size can vary slightly. If you have problems with it fitting, you can install the Arduino Uno boot loader on the Nano to free up more flash space for it to fit. See the Wiki FAQ for details.

closed 2018-02-21 by chamnit

adigitaldj1, 2018-02-21:
You didn't answer all my questions and beside it's not resolved until it is fixed.

chamnit, 2018-02-21:
I did answer all of your questions. All three of them. Please understand that this is not a customer service line. Nobody is paid to help you.

Issues related to common and easy to resolve problems or are not bugs/features related to Grbl itself are closed immediately. We're not here to walk you through everything when there are resources available, like the wiki pages, for you to go through yourself.

Again, I use the Arduino Nano to develop on, so it works just fine. You may need to update the boot loader on it if it doesn't fit. All this is in the Wiki, if you take the time to read it.

chamnit, 2018-02-21:
FYI, here are the direct links to the parts of the wiki you need to know.

[Arduino Nano FAQ](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#grbl-wont-fit-on-my-arduino-nano-or-ftdi-based-arduino-am-i-totally-hosed)
[Compiling Grbl](https://github.com/gnea/grbl/wiki/Compiling-Grbl)

adigitaldj1, 2018-02-21:
Got it THANKS!

Sorry about that post just frustrated.

I did check my communications with another sketch and seems fines and was getting responses to and from.