gnea/grbl #384: grbl and bcnc - no motor activity

crchisholm, 2018-02-21:
I am trying to get started on grbl and bcnc  I have the software installed and it seems to connect to the arduino/grbl and the software "seems" to be performing as I would expect, but nothing is happening at the  motors.   The only thing I have connected to the grbl is the motors and a 12v power supply.  On my other cnc using mach 4/ESS and g540, I know that there has to be either a jumper or a normally closed estop for the motors to run.  Is that the case here?  I can jump it to find out, but I thought there may be other things I need to consider before trying to user the grbl

very sorry if this is not the place to post a question like this.

chamnit, 2018-02-21:
Make sure that your step enable pin is configured to right polarity.

crchisholm, 2018-02-21:
when you say the step enable pin, i assume you mean the enable pin on the stepper drivers correct?  Where exactly do I check that??

crchisholm, 2018-02-23:
I think I understand now.  I need to change $4

crchisholm, 2018-02-25:
Turns out it was not a setting that needed to change.  It was a bad Uno.  Got another and it worked fine.

chamnit, 2018-02-25:
@crchisholm : Great. Glad you got it to work out. Hardware issues can be hard to diagnose and can sometimes be not very obvious. Thanks for reporting.