gnea/grbl #385: UGS not communicating with GRBL board

adigitaldj1, 2018-02-22:
I have an issue with the computer not communicating with the Grbl board, everytime I try to connect from UGS it crashes. I am also not able to get past the $ Help from the Arduino IDE serial monitor. I have installed a different terminal program called Hype Terminal and I am able to connect and communicate with the board. What driver could be missing on my laptop and desktop to cause this issue? Both the laptop and desktop are new and may not have everything needed.

aolshove, 2018-02-25:
I have this same issue.  It's my first time attempting to setup and run GRBL with UGS and I'm using the latest of each.  When I click the port "Open" button to connect to Com4 at 115200, UGS just crashes.  I am also using the latest JRE 9.0.4.  The following error report is generated:
[hs_err_pid5940.log](https://github.com/gnea/grbl/files/1755718/hs_err_pid5940.log)

adigitaldj1, 2018-02-25:
Do away with UGS and download UGS Platform it will then connect and work.

There is setting in Platform you will need to change to run longer lines of Gcode. You will also need to make changes in GRBL firmare settings to get the cnc to make moves what they should be. I ran a 1" circle Gcode and it was only about 3/8". I am going to put a machinist travel indicator on it to get it calibrated. Right now I have it closer then before. I am also going to see if there is a good video on YouTube for UGS Platform.

aolshove, 2018-02-25:
Thanks.  It turned out to be an issue with the JRE 9.0.4.  I uninstalled that and installed Java 8 and it started behaving.  I'll try the UGS Platform as well.

chamnit, 2018-02-25:
@adigitaldj1: UGS is not compatible with Grbl v1.1. There is a wiki page that shows this and other conpatible GUIs. 

You have to configure Grbl for your machine, like any other CNC controller. The wiki configuration page shows you how. 

And, yes, certain versions of the CHxxx USB serial chips do cause streaming problems. Particularly when streaming is near saturated. Please don’t dismiss things as claims when you haven’t yet experienced issues yourself. This is a well documented problem. 

The wiki can be accessed  by clicking the wiki tab present just about everywhere within this site  or the explicit links on the front page.

closed 2018-02-25 by chamnit

winder, 2018-02-25:
This issue is with UGS, not GRBL. There is a bug with the serial library we're using. [Details here](https://github.com/scream3r/java-simple-serial-connector/issues/116).

UGS is fully compatible with Grbl v1.1.

adigitaldj1, 2018-02-25:
Chamnit, Yes agreed that Ugs is not compatible with V1.1 and you should have mentioned that on the posts. All you were worried about was closing the thread. I had to do hours of research to get the machine working. You wanted me to do everything else but not tell me compatibility issues. Gut hub and the Wiki need revised, it is way to hard to find what you are you looking for on these. Yes easy for you because you have been around that but not for a newbie.

I found that UGS Platform works with GRBL V1.1. All of this needs documented on one page! There is also direction I am working on now with a seller for the woodpecker to update their direction on using the the Nano with the CH430G chip to English.

I have to accomplish this all on my own instead of being lead down the wrong road by flashing and changing this and that etc. when it wasn't the issue. Like I said before, making sure to have the jsva jre up to date, flash v1.1, do away with UGS and install UGS Platform and that will get it working. I haven't said anything else about getting other issues and I'll deal with at along the way!

adigitaldj1, 2018-02-25:
@Will,

I could not get UGS to work with V1.1 until I went to UGS Platform.

winder, 2018-02-25:
@adigitaldj1 please feel free to open UGS related issues (such as this one) on the [UGS GitHub page](https://github.com/winder/Universal-G-Code-Sender/issues). In this case it is my fault, because I don't release UGS very often. There are dozens and dozens of GRBL front ends now and @chamnit can't be expected to support all of them!

adigitaldj1, 2018-02-25:
In the beginning I had no idea what the issue was. If it was an issue with GRBl, nanao, ch430 or UGS. Then I ran Hype Terminal and was able to connect. So that told me my first problem may have been UGS. Anyway farther ahead then I was a couple of days ago.

chamnit, 2018-02-25:
@winder : My bad. I thought the old UGS wasn’t compatible and only UGS Platform was. Thanks for the update. 

@adigitaldj1 : I’m sorry you had so many issues, but based on all past questions, none of the issues were problems with Grbl in particular. Hence your threads were closed. As you found out, it was an incompatible or buggy version of UGS and reading the Wiki that was the problem. 

Please try to understand that we get lots of newbie questions weekly and almost daily. Many of them are due to users not reading the Wiki or not have bothered learning anything about CNC and expecting someone to walk them through everything. After a while, that gets very exhausting. 

You’re right though. I thought there was a FAQ answer that told people that v1.1 requires a GUI updated with to the new protocol, but there are only a few scattered mentions of it. I just added a new FAQ to help keep this from happening again.

will, 2018-02-25:
hugs not ugs

adigitaldj1, 2018-02-27:
I truly appreciate everything you guys do! I wasn't frustrated at you just the lack of some information out there and seemed like I was the only one with the particular trouble. I beieve I have everything working good now and just fihuring out getting this all setup to run some panels for some meters I make.

I had a friend run a couple of the meters on his machine and he gave me the code, but is was done with conversation on a Milltronics controller. I am not sure what I need to get rid of yet in the gcode to run on grbl.

Is there any program out there can strip out what I don't need or would it be easier to goto to the drawing board(Fusion 360) and just draw it up then send it out the gcode or .nc?