gnea/grbl #390: Compiling issue

knightofkairell, 2018-03-04:
I have been trying to get either the stock version or the config file modified for CoreXY version to compile. But it keeps throwing errors. 

Required HOMING_CYCLE_0 not defined.
WCO refresh must be greater than one.
Override refresh must be greater than zero.

I have defined the first error, but it still give the code. What am I doing wrong? I want to compile with only turning on CoreXY. But I can't even get it to compile without changing anything. Still gives me the same error codes.

knightofkairell, 2018-03-04:
[image]

knightofkairell, 2018-03-04:
Even tried on Windows with latest 1.8.5 Arduino IDE. Same 3 errors.

JBSchueler, 2018-03-04:
Looking into `config.h` you have to comment line `106` and `107` and uncomment line `113` and `114` to solve your `HOMING_CYCLE_0` issue. The problem is that at least `HOMING_CYCLE_0` must be defined. What it defines is up to you. But as can be read in the code, homing X and Y simultaneously seems not to be compatible.

```c
//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
```
```c
 #define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
 #define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y
```

The other 2 errors are related to `REPORT_WCO_REFRESH_IDLE_COUNT`  and `REPORT_OVR_REFRESH_IDLE_COUNT`.
```c
#if (REPORT_WCO_REFRESH_IDLE_COUNT < 2)
  #error "WCO refresh must be greater than one."
#endif
#if (REPORT_OVR_REFRESH_IDLE_COUNT < 1)
  #error "Override refresh must be greater than zero."
#endif
```
These values are also defined in `config.h` at line `286`.
```c
#define REPORT_OVR_REFRESH_BUSY_COUNT 20  // (1-255)
#define REPORT_OVR_REFRESH_IDLE_COUNT 10  // (1-255) Must be less than or equal to the busy count
#define REPORT_WCO_REFRESH_BUSY_COUNT 30  // (2-255)
#define REPORT_WCO_REFRESH_IDLE_COUNT 10  // (2-255) Must be less than or equal to the busy count
```
If you didn't changes these setting, the defines are greater than `0` and `1` so that shouldn't be a problem.
Are you sure you didn't change these values?

knightofkairell, 2018-03-04:
Those are my values for ovr and wco refresh, didn't change them from the original. And I did the exact same thing you did for homing CoreXY. Plus the line thats says define CoreXY. And that was all I changed. But it didn't work. Neither did the original, if I change nothing it still doesn't work.

cojarbi, 2018-03-04:
Also having the same issues as @knightofkairell . I did this in the past with previous grbl version and compiled without issue by just uncommenting #define corexy

@chamnit any suggestion or can you point what are we doing wrong here? Thanks

chamnit, 2018-03-04:
This is a very common problem. It’s due to not following the Wiki compiling instructions. There is a warning about removing all previous Grbl installations in the Arduino library paths. If you don’t, this compile error happens.

closed 2018-03-04 by chamnit

knightofkairell, 2018-03-04:
@chamnit I followed the instructions to the letter. And tried on two different computers.

chamnit, 2018-03-04:
@knightofkairell : For the sake of argument, I followed the wiki compile steps with a fresh download of the Arduino IDE 1.8.5 (and 1.8.4) on Mac. Imported the grbl library from a fresh unaltered ZIP download from the Github site. Altered only the config.h that was now copied into the Arduino library and uncommented the COREXY line. Used the GrblUpload example with no alterations, as it was stated in its comments. It compiles fine. 

Again, follow the directions exactly. Do not alter anything except the config.h file.

chamnit, 2018-03-04:
BTW, Grbl with CoreXY enabled will not fit on an Arduino Nano, as is. You will need to update the bootloader to the Arduino Uno's to free up another 1.5KB of flash. After you do so, your Nano will be an "Arduino Uno" in terms of the Arduino IDE and, if flashing times out, it may require a timely reset right as it starts to flash. It's easy to do and can be switch back easily.

knightofkairell, 2018-03-04:
Ok so the problem is adding the grbl library to the grblUpload sketch. My mistake, sorry. I apperantly either missed it or ignored the "Do not alter this file!" statement. Sorry. 

Now about changing the bootloader for the Nano. what programmer should I select. I am guessing that to burn the Uno bootloader I have to select board, Arduino/Genuino Uno. Select processor ATmega328P. Select the port that is the current Nano. But which programmer do I select before clicking burn?

knightofkairell, 2018-03-04:
Also should I be seeing a bunch of redefine statements in orange once it compiles. None of them say error. It does say it is done compiling and that it will use 32168 bytes which is 99% of storage space of the Uno. And 1633 bytes which is 79% of dynamic memory. That sound right?

chamnit, 2018-03-04:
[Click the link](https://github.com/gnea/grbl/wiki/Compiling-Grbl#via-the-arduino-ide-all-platforms-recommended-for-all-users) on the last note. 

No redefine messages should be present if done right.

99% full flash for CoreXY when compiling on an Arduino Uno.

knightofkairell, 2018-03-04:
Thanks I got it working. Sorry about the trouble.

miro-ux, 2022-03-08:
I got this error when trying to upload the from **Sketch>Include Library>grbl** as opposed to the proper way: **File>Examples>grbl>grblUpload**. Rookie mistake 😳