gnea/grbl #394: How to convert simple_stream.py code to C#

Rob2029, 2018-03-09:
Hello everyone! i am working on a interface to control my cnc using C# currently i am trying to send a gcode to my machine and i found this method (https://github.com/grbl/grbl/blob/master/doc/script/simple_stream.py)
I undestand the code and how it works but i can´t tranlate it to C# quite right. If anyone knows how to do it i would love to hear it!
Greetings from Costa Rica

chamnit, 2018-03-09:
The [Grbl Interface](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface) is described here. You'll have to figure out how to convert the streaming code to C# yourself. It's all commented, straight forward, and self-explanatory.

closed 2018-03-09 by chamnit

zeevy, 2018-03-10:
Have a look at https://github.com/arkypita/LaserGRBL

Rob2029, 2018-03-10:
Thanks again Zeevy, btw do you know any code converter from phyton to C#?

zeevy, 2018-03-10:
As far as i know there is no such automated process.
I think that LaserGRBL is written in C#, may be you can use some portion of the code.

Rob2029, 2018-03-10:
Thanks i'll check it out