gnea/grbl #395: grbl v1.1 x axies problem

ikram7306, 2018-03-11:
when i upload g code using universal gcode sender then my grbl cnc z axis moves down milling position at the same time z axis start moving upward and stop to that position where it starts at the begining.
i using fusion 360 for cad and cam
my g code is
 %
(1001)
(T1  D=3.175 CR=0 - ZMIN=-1 - flat end mill)
G90 G94
G17
G21
G28 G91 Z0
G90

(Face1)
M9
T1 M6
S10000 M3
G54
M8
G0 X17.064 Y-8.697
Z16
Z5
G1 Z-0.683 F762
G18 G3 X16.746 Z-1 I-0.317
G1 X15
X-15
G17 G2 Y-5.838 J1.43
G1 X15
G3 Y-2.978 J1.43
G1 X-15
G2 Y-0.118 J1.43
G1 X15
G3 Y2.742 J1.43
G1 X-15
G2 Y5.602 J1.43
G1 X15
G3 Y8.462 J1.43
G1 X-15
G18 G3 X-15.318 Z-0.683 K0.317
G0 Z16
G17
M9
G28 G91 Z0
G28 X0 Y0
M30
%

jahnj0584, 2018-03-11:
You have to disable the G28 option in your cams post processor. Thank me
later

biasedlogic, 2018-03-12:
> You have to disable the G28 option in your cams post processor. 

No, not really... It's better to configure it properly. Your machine needs to have homing enabled and homing switches set up properly, then you set up the safe position (G28) at top of the milling space and then the G28 command does what it's supposed to: retracts the tool to safe height before firing up the spindle.

ikram7306, 2018-03-12:
Please tell me how i configure prooer homing
Thank you very much for helping sir.......

chamnit, 2018-03-12:
@ikram7306 : Read the configuration page in the [wiki](https://github.com/gnea/grbl/wiki) on how to configure Grbl.

closed 2018-03-12 by chamnit

jahnj0584, 2018-03-12:
Yes but you have no need to. G28 goes to machine home. This can be a long
way from your part zero and is just a time waster. People who have this
issue are those without hard limits and homing cycles.

biasedlogic, 2018-03-12:
> G28 goes to machine home.

1. G28 G91 Z0 moves Z-axis only, it does not move XY axes.
2. Where does this command send your unit, can be set using G28.1 and does not have to be home zero position.

> People who have this issue are those without hard limits and homing cycles.

Which is what I'd consider a root problem of many other problems, why I also wrote that you have to have reference+homing working first...

regards

chamnit, 2018-03-12:
@biasedlogic : For Grbl and LinuxCNC, G28 does not go to machine home. It goes to a predefined positions, as shown [here](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g28-g28.1). Also, I agree G28 (and G30) are not useful unless you have homing enabled, since the predefined position references the home position.

biasedlogic, 2018-03-12:
@chamnit

>@biasedlogic : For Grbl and LinuxCNC, G28 does not go to machine home. It goes to a predefined position, as shown here.

Isn't it what I have just said?

chamnit, 2018-03-12:
@biasedlogic : lol. You sure did. Not sure where my reading comprehension skills are today.

langwadt, 2018-03-12:
you can't really use G28 without setting a position with G28.1 (or building with HOMING_FORCE_SET_ORIGIN) because going to 0 will hit the limit switches 

@chamnit wouldn't it make more sense if 0,0,0 was a pull of distance from the switches by default?

biasedlogic, 2018-03-12:
> wouldn't it make more sense if 0,0,0 was a pull of distance from the switches by default?

I'm not chamnit, but I'd vote against it. Zero is zero, pull off can be tweaked later, there's reason for the ability to set g28 reference, it's good side effect to force it

chamnit, 2018-03-12:
@langwadt : In the next major version, users will have greater control how the homing origin is set, among other things. For Grbl 328, it's going to stay as is, mainly for simplicity and not having to ask GUIs to accommodate new stuff with Grbl's limited space to add more feedback.

ikram7306, 2018-03-14:
i using universal g code sender v 1.0.9 and grbl v1.1
i put the limit switches with cnc then i have a problem

**** Connection closed ****
**** Connected to COM9 @ 115200 baud ****

Grbl 1.1f ['$' for help]
ALARM:1
[MSG:Reset to continue]
>>> G91 G0  Y1

then i press Y+ my stepper not start..........