gnea/grbl #396: Servo control

joepuzzo, 2018-03-13:
So i am building a two axis grinder and one axis uses a linear actuator with servo. Our goal was to control the two axis using gcode with grbl. i wonder if anyone has created a coverter from stepper to pwm. ( obviously with some loss in accuracy ). But it would be slick if i could send gcode commands out via grbl anc control servo for x or y axis. And yes i have already looked at grbl servo and it only allows pwm control for M3 command :( Any ideas on how i could solve this

shooter64738, 2018-03-13:
Could you use a servo drive, which would read the pulse and direction data from grbl and drive your servo that way?
Optionally, use a separate avr board to read the pulse data on one port, and output a pwm signal on another port independent of grbl? You would need a dc motor driver (ebay for $10) to run your servo. 

Im not entirely sure what you want to do, but I think one of those might do it without reworking pieces of grbl.

joepuzzo, 2018-03-13:
@shooter64738  Do you happen to know where i could find one of those separate avr boards? Basically the goal is to make the servo an x or y axis.. so when . grbl gets `X10` it will drive a servo instead of a stepper.

joepuzzo, 2018-03-13:
Oh just realized AVR is just a generic term for type of microcontroller lol. So i would have to write some custom logic on a board to take in stepper input and go to servo?

shooter64738, 2018-03-13:
The cheapest thing is probably an arduino uno. Although you can get a  mega2560 on ebay for $10 or so. Ebay has been my go to source for arduino/avr boards for a while. :)

joepuzzo, 2018-03-13:
LOL i already have like 10 arduinos and a bunch of teensys laying around so i can probs use one those :) Was trying to be lazy and not have to write code to transfer Stepper pulses ---> PWM. Figured someone had to have done that already.

shooter64738, 2018-03-13:
I drive a servo as a spindle, and although its a separate board from grbl, I use the pwm signal to drive a dc motor controller about to 96 volts and 20 amps. I dont drive it right off the arduino. But in your case to convert that signal you will see pulses that just increase with frequency, starting at 0 and going to some unknown value. As grbl accelerates the axis the pulses will become faster and faster. So to know when you should be at 100% pwm you will either have to calculate it based on the max pulse rate, or just fiddle with it until you get to the speed you want it to run at. 

Im sure someone probably has done something like this for something out there. This is pretty much what a servo driver does. Converts pulse and direction logic into an analog signal to drive a big servo. Along with a few other things. But they can be pretty pricey for a big one.

joepuzzo, 2018-03-13:
Right that makes sense.. as long as i can map stepper pulses and dir ----> PWM value 0-255 i should be all set.

ASHIT-AXAR, 2018-03-25:
there are some reasons to not doing that.
- there are no more space left on the chip.
- there are no benefits in doing Servo Control.
    -- high quality CNC machines using Servo motors : those are already using (step/direction) logic by setting Servo Driver(s have 3 modes) in that mode.
    -- its wasting time to use those low (quality, speed) and inaccurate small Servos in low budget CNC machines. you gaining no benefits over stepper motors.

joepuzzo, 2018-03-25:
I found Clear paths stepper killer motors so all is well. They take step and direction signals! https://www.teknic.com/products/clearpath-brushless-dc-servo-motors/clearpath-sd-stepper-replacement/

closed 2018-03-25 by joepuzzo

X3msnake, 2018-03-25:
> products/clearpath-brushless-dc-servo-motors/clearpath-sd-
> stepper-replacement/

Cool post back with your experience on those if you end up using them.