gnea/grbl #397: GRBL mistreats cancelled jobs

jahnj0584, 2018-03-14:


chamnit, 2018-03-14:
It’s not clear what you are referring to as a problem. You seem to probe, play with overrides, then cancel the job. I see the z value change but don’t see what happens to the machine. It doesn’t sound like anything happens.

jahnj0584, 2018-03-14:
Cancelling a job after running a probe cycle causes the probed axis'
current location to change. G53 remains the same if memory serves correct.
The over rides where to just show that it was moving beforehand anyways.

chamnit, 2018-03-14:
So after a probe cycle? The dynamic tool length offset was likely appropriately cleared.

jahnj0584, 2018-03-14:
Why would the offset be changed when a job is cancelled but remain when the
job is sent to completion?

chamnit, 2018-03-14:
It depends on what UGS does when the job is cancelled. If it automatically soft resets Grbl, it’ll clear all temporary offsets. 

As you said G53 or machine coordinates are fine, only the work position shifts. This means something in the work offsets is altered. Again likely the tool length offset. Grbl doesn’t do anything to that offset unless there is a reset. If you print the $# parameters and $g gcode state before and after, you can see if anything changed.

closed 2018-03-14 by chamnit

biasedlogic, 2018-03-15:
> Why would the offset be changed when a job is cancelled but remain when the job is sent to completion?

@Chamnit told you why does that happen, as in 'what leads to the effect', as to the reason: reset is a state that has no known duration, grbl can't know why it was in reset, how long and what happened there. Note, that homing is assumed lost too.

jahnj0584, 2018-03-18:
If I use a G10 P0 L20 Z14.91 instead of G92 Z14.92 it should work and stay after a reset, right? Its the same as hitting the reset Z zero button.

chamnit, 2018-03-18:
G10 saves to eeprom so yes. G92 is defined as temporary in the gcode standard. So it resets with a reset.

jahnj0584, 2018-03-18:
Great. I'll just so that then. Why do people suggest G92 then?