gnea/grbl #398: Possibility to invert Spindle Direction PIN?

saseba, 2018-03-17:
Hello,

is there a Possibility to invert the behavior of the Spindle Direction PIN (D13)? 
GRBL offers this option for the Spindle Enable and the Coolant pin, but not for the Spindle Direction pin, or did I miss something?

In spindle_control.h, I tried to simply switch:
#define SPINDLE_STATE_CW       bit(0)
#define SPINDLE_STATE_CCW      bit(1)
to
#define SPINDLE_STATE_CW       bit(1)
#define SPINDLE_STATE_CCW      bit(0)

but this doesn't seem to do the trick.

config.h:
...
// Inverts the spindle enable pin from low-disabled/high-enabled to low-enabled/high-disabled. Useful
// for some pre-built electronic boards.
// NOTE: If VARIABLE_SPINDLE is enabled(default), this option has no effect as the PWM output and
// spindle enable are combined to one pin. If you need both this option and spindle speed PWM,
// uncomment the config option USE_SPINDLE_DIR_AS_ENABLE_PIN below.
#define INVERT_SPINDLE_ENABLE_PIN // Default disabled. Uncomment to enable.

// Inverts the selected coolant pin from low-disabled/high-enabled to low-enabled/high-disabled. Useful
// for some pre-built electronic boards.
#define INVERT_COOLANT_FLOOD_PIN // Default disabled. Uncomment to enable.
// #define INVERT_COOLANT_MIST_PIN // Default disabled. Note: Enable M7 mist coolant in config.h

langwadt, 2018-03-17:
can always just edit spindle_control.c

chamnit, 2018-03-23:
It's easy to edit the spindle_control.c file. Just a few lines of code that you need to swap. But this is already planned to be configurable in the next major version.

closed 2018-03-23 by chamnit

JamieAKing, 2019-05-06:
my vfd has analog input for spindle speed which works fine but the enable function requires a 15v pin to be pulled to ground. I can do this through a relay but will need a spindle enable pin at 5v on the Arduino to trigger the relay. is anyone willing to point at exactly which lines of code and the changes needed to use spindle direction(d13) as a seperate spindle enable while keeping the Pmw speed control on pin 11

pc-maniac, 2019-12-01:
> my vfd has analog input for spindle speed which works fine but the enable function requires a 15v pin to be pulled to ground. I can do this through a relay but will need a spindle enable pin at 5v [...]

config.h
define USE_SPINDLE_DIR_AS_ENABLE_PIN