gnea/grbl #399: No status in second route

momo-aux, 2018-03-18:
Hi, using flask and pyserial I can't get the status command to work (b'?') in a second route.
It works if I add it to the end of the / route but I want to do ajax calls on this to update the work coordinates on the page.

If you have any hints I would love to hear them.

--- snip ---
ser = serial.Serial('/dev/ttyUSB0',115200)
ser.write(b"\r\n\r\n")
time.sleep(2)
ser.flushInput()
ser.write(b"$X\n")

--- snip ---

@app.route("/", methods=['POST', 'GET'])
def index():
    if 'homing' in request.form:
        ser.write(b'$H\n')
    return render_template('index.html')

@app.route('/status')
def status():
    global ser
    ser.flushInput()
    ser.write(b'?')
    return 'Status: {0}'.format(ser.readline())

chamnit, 2018-03-18:
@momo-aux : Not a Grbl issue. Closing this thread, but people are still able to respond and help. I'd try to help, but it's a bit vague what your code is doing and what you'd like for it to do.

closed 2018-03-18 by chamnit

momo-aux, 2018-03-18:
I use flask to create a simple web app.
Just some buttons to send a gcode to grbl via pyserial. 
I want to read the status from JavaScript via Ajax with a get to the flask app on route /status
The guide sending is working fine.
But serial.write(b'?') and then a ser.readline() only returns blank string. No < string with mpos and wpos.

momo-aux, 2018-03-18:
guide=gcode autocorrect